libretro/Lakka-LibreELEC #1151: GPIOs Controller not working on Nightly builds

rafaelmartines, 2020-12-30:
**- Which version of Lakka are you using?**
I usued the last image available for RPi2/3, Lakka-RPi2.arm-nightly-20201020-324ceda.img.gz
**- What system hardware are you using?**
RPi3 B+
**- What did you do?**
I tried to follow this tutorial to get the GPIO Controllers working: https://github.com/Turro75/Lakka-LibreELEC/wiki/Generic-GPIO-controller-on-lakka-2.3.2
- What did you expect to happen?
On Lakka 2.3.2 when you bootup after doing the tutorial steps, both GPIOs controllers are recognized on retroarch screen, but on nightly buids is not, I tried both methods, by upgrading and by doing a fresh install
- What happened instead?
The GPIO Controllers are not recognized before and after the tutorial on Nightly builds.

Turro75, 2021-01-02:
Hello,

Are You able to connect to the lakka system by ssh and verify if the module mk_arcade_joystick_pi exists and can be loaded?
Can You share the output of the command dmesg?

Turro75, 2021-01-03:
It seems the standard build excluded the driver, so I think You have to compile yourself lakka 3.
I just recompiled the driver for Your target, it should work.[https://drive.google.com/file/d/1zOUVKF9Zt6FD1ABxGk0I0tHnoNhYOfbS/view?usp=sharing](https://drive.google.com/file/d/1zOUVKF9Zt6FD1ABxGk0I0tHnoNhYOfbS/view?usp=sharing)

rafaelmartines, 2021-01-04:
OMG! I didn't get the notification, I mighy try it today after work, thank you :D

rafaelmartines, 2021-01-04:
> Hello,
> 
> try following this tutorial:
> https://github.com/Turro75/mk_arcade_joystick_rpi/wiki/Raspberry-pi-and-GPIO-controllers-with-Lakka-3
> 
> Are You able to connect to the lakka system by ssh and verify if the module mk_arcade_joystick_pi exists and can be loaded?
> Can You share the output of the command dmesg?

Thanks dude, you can now close the issue, I didn't need to excluded driver on the nighly build available for RPi2/3, following your tutorial worked like a charm

rafaelmartines, 2021-06-10:
Hi @Turro75 Did you manage to make GPIO controller on the official (or recent) release of Lakka 3?

rafaelmartines, 2021-06-10:
@ToKe79 I'll test it later after work

rafaelmartines, 2021-06-10:
Hi @ToKe79 I just booted the image and followed the tutorial to activate the GPIO controllers and it worked :D

I'll close it :smile: :pray: thank you :heart:

commit 447e58c8 (2021-06-10): RPi: enable wii-u-gc-adapter and mk_arcade_joystick on all RPi devices except GPICase (#1386)

As RPi3.aarch64 was added as separate platform, these packages are not
included. As we have dropped support for Gamegirl, it is easier to make
the condition DEVICE is not GPICase, as all other RPi devices should
include these packages

Fixes #1151
closed 2021-06-10 by rafaelmartines