libretro/Lakka-LibreELEC #1155: Additional MAME cores

spamsool, 2021-01-05:
Hi

Could you please add additional MAME cores to nightly 32bit x86 build of Lakka? From my MAME roms collection Lakka detects only MAME 2003-Plus and MAME 2010 roms. And there are many more to add like MAME 2000, MAME 2003, MAME 2015 and MAME 2016.

Would you be so kind to add those? Thanks a lot in advance.

I am using nightly x86 build on my old 32 bit only core duo laptop with intel gma950 and 1280x768 screen resolution  because on the stable build I get no picture on the screen at all.

Thanks :)

Ntemis, 2021-01-05:
That's not possible as we had to exclude some mame cores due to space limitations. Why you need every mame core in existence if i may ask.

spamsool, 2021-01-06:
Hi
You might be right - space limitations sounds like a reasonable reason. I wantes more cores to be able to run more mame games, which I think often require a specific core but I can live with that. So we might as well leave it as it is now :) Thanks for the good work guys anyway.

closed 2021-01-06 by spamsool