libretro/Lakka-LibreELEC #1156: RetroELEC support for the new ODROID-Go Super by Hardkernel

Hedda, 2021-01-13:
Hardkernel has announced ODROID-Go Super a few weeks ago and I believe that they offer samples to OS developers like you?

https://forum.odroid.com/viewtopic.php?f=187&t=41283&p=315650

It basically has the same internals as ODROID-Go Advance but comes with 5-inch 16:9 (FWVGA) square-pixel aspect radio screen with 854 x 480 pixels resolution. It is also an offline device without built-in WiFi but users can temporarily use a USB WiFi dongle.

https://github.com/hardkernel/linux/commits/odroidgoA-4.4.y

https://github.com/hardkernel/u-boot/commits/odroidgoA-v2017.09

Ntemis, 2021-01-13:
Yes it's time to add support. Thanks for reminding me.

h1aji, 2021-02-23:
Would be very nice to see Lakka running on OGS @Ntemis

h1aji, 2021-02-25:
Patches could be ported from this commit: https://github.com/EmuELEC/EmuELEC/commit/8e3f73cbb865b2ddcbfa91949092723d6a93405e

h1aji, 2021-02-27:
@Ntemis thanks for merging my PR, I have noticed that nightly build failed. Is there any chance to build 
`PROJECT=Rockchip DEVICE=OdroidGoAdvance ARCH=aarch64 make image`
instead of ARCH=arm

Thanks

h1aji, 2021-03-04:
I have added a few more commits in this PR https://github.com/libretro/Lakka-LibreELEC/pull/1192
This one is going to fix previously broken Rockchip builds

davidgfnet, 2021-03-20:
This should be now fixed.
Sent an announcement to the odroid forum: https://forum.odroid.com/viewtopic.php?f=193&t=42159

closed 2021-03-20 by davidgfnet