libretro/Lakka-LibreELEC #1161: Travado na flor de inicialização com reboot

mbmoreira84, 2021-01-16:
- Qual versão do Lakka você está usando?
- não sei, técnico de informática que fez pra mim.
 - Qual hardware de sistema você está usando?
 - PC Linux Dell Inspiron 1525 2gb dualcore
 - O que você fez?
 - Acessar inclusão de emulador
 - O que você esperava que fosse acontecer?
 - abrir pasta de emuladores
 - O que aconteceu em vez disso?
 - travando na flor da inicialização com reboot.

Ntemis, 2021-01-16:
English please

mbmoreira84, 2021-01-16:
What version of Lakka are you using?

I don't know, computer technician who did it for me.

What system hardware are you using?

Dell Inspiron 1525 2gb dualcore Linux PC

What did you do?

Access emulator inclusion

What did you expect to happen?

open emulators folder

What happened instead?

locking on the boot flower with reboot.

Obter o Outlook para Android<https://aka.ms/ghei36>

Ntemis, 2021-01-16:
Try this build https://nightly.builds.lakka.tv/2021-01-15/Lakka-Generic.x86_64-3.0-devel-20210115230201-6f20233.img.gz

mbmoreira84, 2021-01-16:
Thanks.  I give information about the procedure.

Obter o Outlook para Android<https://aka.ms/ghei36>

gouchi, 2021-04-21:
Hi,

Did you have some opportunity to make some tests otherwise we will close this issue ?

Thank you.

closed 2021-07-13 by ToKe79

mbmoreira84, 2021-07-13:
Good Morning.  I haven't had the opportunity to work on fixing the problem yet.  Thanks for listening.

Obter o Outlook para Android<https://aka.ms/AAb9ysg>

Closed #1161<https://na01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Flibretro%2FLakka-LibreELEC%2Fissues%2F1161&data=04%7C01%7C%7Cf3ad1f61b3f8447bd56108d945f4fe17%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637617740800500254%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C1000&sdata=5LrEJXiKCLr9zQcbX0lfNTuoAF2BDWoFOOEYzqcY6rs%3D&reserved=0>.