libretro/Lakka-LibreELEC #1166: ODROID-XU4 not booting Lakka nightly

darklordtom, 2021-01-29:
Hi,

I'm having a problem to install latest nightly on Odroid XU4 emmc. I'm booting minimal ubuntu from SD and using dd to flash the image to emmc:
dd if=Lakka-OdroidXU3.arm-nightly-20210122-45c6f56.img of=/dev/mmcblk0 bs=4M conv=fsync
Process finishes without obvious errors.

**When I restart Odroid with jumper=emmc, I can see lakka logo(flower) and cursor blinking in upper left corner also getting the blue light flashing with sequence: blink,blink pause repeated. Nothing happens after that, waited for 10mins**

When I restart Odroid with jumper=sd and look at emmc I can see:
root@odroid:~# ls /dev/mmcblk0*
/dev/mmcblk0  /dev/mmcblk0boot0  /dev/mmcblk0boot1  /dev/mmcblk0p1  /dev/mmcblk0p2  /dev/mmcblk0rpmb

root@odroid:~# fdisk -l
Disk /dev/mmcblk0: 58.25 GiB, 62537072640 bytes, 122142720 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0xda294b4e

Device         Boot   Start     End Sectors  Size Id Type
/dev/mmcblk0p1 *       3072 1051647 1048576  512M  c W95 FAT32 (LBA)
/dev/mmcblk0p2      1051648 1117183   65536   32M 83 Linux

When I mount the Linux partition is empty and quite small
When I mount the fat32 partition, it's having expected content:
9eaa46e9ceb7695e447c07c907cf553a  target/KERNEL
root@odroid:/media/emmc# md5sum KERNEL
9eaa46e9ceb7695e447c07c907cf553a  KERNEL
root@odroid:/media/emmc# cat SYSTEM.md5
6b2030c9e5f66b6873598710887a1aee  target/SYSTEM
root@odroid:/media/emmc# md5sum SYSTEM
6b2030c9e5f66b6873598710887a1aee  SYSTEM

Tried to resize the second partition via:
parted
resizepart 2
-1
 e2fsck -f /dev/mmcblk0p2
resize2fs /dev/mmcblk0p2

Tried to format 
mkfs.ext4 /dev/sdb1

But it didn't help

Ntemis, 2021-01-29:
I confirm your issue. We are unbootable on xu3&4. I will try to fix it and post an image for you to test next week

Ntemis, 2021-06-06:
Please test 3.1 release

closed 2021-06-20 by ToKe79