libretro/Lakka-LibreELEC #1167: [Feature request][Development]: Firmware packages for rtl8723cs and anx7688

psstoyanov, 2021-02-03:
Hi, I'm progressing with getting Lakka working on the PinePhone but I got into a stumbling block.

I'm not sure how to create the required firmware packages for `rtl8723cs` and `anx7688`.

So far I've based my changes on Danct's Arch linux for the kernel which can be found here: https://github.com/dreemurrs-embedded/Pine64-Arch/tree/master/PKGBUILDS/pine64 
It works well however the WiFi/Bluetooth and USB-C port aren't functioning as expected.

I've set the PinePhone as separate Allwinner target while attempting to minimise any changes to already existing code.
My fork and changes can be found here: https://github.com/psstoyanov/Lakka-LibreELEC/tree/pinephone_test

The problem which I'm facing is in the creation of the firmware packages themselves. How would the `package.mk` have to look like? 
The current `rtl8723bt` doesn't appear to have anything inside it's body apart from the sources. Would the same be true for the `rtl8723cs` ?

gouchi, 2021-02-04:
Hi,

Thank you for work, looking forward to seeing Lakka running on the PinePhone ;)

According to [pmOS](https://gitlab.com/postmarketOS/pmaports/-/blob/master/device/main/device-pine64-pinephone/APKBUILD#L130-L133) you need from [linux firmwares](https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/) : [rtlwifi](https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/rtlwifi) and [rtl_bt](https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/rtl_bt).  So you should [add them with a condition in linux package](https://github.com/psstoyanov/Lakka-LibreELEC/blob/pinephone_test/packages/linux/package.mk#L168-L170) and enable in the [kernel config](https://github.com/psstoyanov/Lakka-LibreELEC/tree/pinephone_test/projects/Allwinner/linux) [CONFIG_EXTRA_FIRMWARE and CONFIG_EXTRA_FIRMWARE_DIR](https://github.com/psstoyanov/Lakka-LibreELEC/blob/pinephone_test/projects/Generic/linux/linux.x86_64.conf#L1512-L1513).

For "outside" linux firmware, you should add the package _firmware-pinephone_ in [linux-firmware](https://github.com/psstoyanov/Lakka-LibreELEC/tree/pinephone_test/packages/linux-firmware) directory and add it [here](https://github.com/psstoyanov/Lakka-LibreELEC/blob/pinephone_test/projects/Allwinner/devices/PinePhone/options). Please check for example [firmware-dragonboard](https://github.com/libretro/Lakka-LibreELEC/search?q=firmware-dragonboard).

They added [rtl8723bt](https://gitlab.com/postmarketOS/pmaports/-/blob/master/device/main/firmware-pine64-rtl8723bt/APKBUILD), from this [repository](https://github.com/anarsoul/rtl8723bt-firmware).

So I don't know if `rtl8723cs` and `anx7688` are needed but you can add [them](https://megous.com/git/linux-firmware/tree/) for your device  ?

gouchi, 2021-04-21:
Hi,

Did you have some success ? ;)

Thank you.

psstoyanov, 2021-04-22:
Hi, @gouchi , I haven't worked on this for awhile now. I couldn't get the WiFi and Bluetooth to work at time.
It took a back seat once we managed to get feedback from the serial console/ usb-c ethernet and then my compile problems with Wayland (partially my systems fault as it wants updated packages and partially the meson build system - I've messed somewhere the targets for cross-compiling).

I have a couple of classes to catch up with for this weekend so my plan is to come back to this next Tuesday or Wednesday.

gouchi, 2021-04-23:
Hi @psstoyanov,

Thank you for the update, no rush ;)

Ntemis, 2022-06-10:
This appears dead

closed 2022-06-10 by Ntemis