libretro/Lakka-LibreELEC #1174: Pi4 gpio-shutdown not working

donald-keighley, 2021-02-07:
Pi4 8GB model with Lakka-RPi4.arm-nightly-20210206-4ea79b3.img.

Using this in config.txt to power on and shutdown with a button as well as turn on the button's led.: 

`dtoverlay=gpio-shutdown,gpio_pin=3
dtoverlay=pi3-act-led,gpio=23
dtparam=act_led_trigger=default-on`

This works with raspbian and it used to work on the stable version of Lakka, so it's not a problem with how the button is wired.  The led part works with the current nightly, but the shutdown part does not.

Turro75, 2021-02-15:
I had similar issues while porting gpio joystick controls to pi4 due to changes in how gpio is initialized, while fixing the code I used to set input pullups in the config.txt as reported here: https://www.raspberrypi.org/documentation/configuration/config-txt/gpio.md

gouchi, 2021-04-21:
It is it still an issue with [latest nightly](http://nightly.builds.lakka.tv/latest/) ? Moreoever, you can use now [RPi.GPIO](https://github.com/libretro/Lakka-LibreELEC/pull/1264).

Thank you.

jmdcampos, 2021-04-21:
@gouchi , this dtoverlay line still does not work. For now, RPi.GPIO is the way to go. 
However, using config.txt is much easier and probably also requires less CPU cycles than a Python script.

Using Lakka-RPi4.arm-nightly-20210419-5a4fc0e.img

donald-keighley, 2021-04-24:
Just tried the latest nightly Lakka-RPi4.arm-nightly-20210423-d52f851.img, still did not work.

donald-keighley, 2021-05-22:
Still not working in Lakka 3.0 for Pi4.

donald-keighley, 2021-05-23:
Based on this [post](https://www.raspberrypi.org/forums/viewtopic.php?t=261452) it looks like 

```HandlePowerKey = ignore```

Should be changed to 

```HandlePowerKey = poweroff```

in the file /etc/systemd/logind.conf.

Unfortunately, I can't edit it because it's read only.

gouchi, 2021-05-23:
@donald-keighley 
Thank you for the investigation.

It seems to come from this [commit](https://github.com/libretro/Lakka-LibreELEC/commit/bf217bc814ba5489b3208019d0559717d2b5b1b9).

According to the [documentation](https://www.freedesktop.org/software/systemd/man/logind.conf.html),

> Controls how logind shall handle the system power, reboot and sleep keys and the lid switch to trigger actions such as system power-off, reboot or suspend.
> ...
>  If "ignore", logind will never handle these keys.

So we may change to `poweroff`, @dmrlawson do you concur ? ;)

dmrlawson, 2021-05-23:
> @donald-keighley
> Thank you for the investigation.
>
> It seems to come from this [commit](https://github.com/libretro/Lakka-LibreELEC/commit/bf217bc814ba5489b3208019d0559717d2b5b1b9).
>
> According to the [...]

Sounds good to me. This will affect every version though, not just pi4? Any idea what the motivation was for changing it?

gouchi, 2021-05-23:
We may try to push it only for RPi at first ?

I don't know why they did like that for upstream LE.

donald-keighley, 2021-05-23:
Not sure why it's set to ignore but I can test it out if you change it.

donald-keighley, 2021-05-24:
@dmrlawson and @gouchi it works in the latest nightly!  Thanks so much.

closed 2021-05-24 by ToKe79