libretro/Lakka-LibreELEC #1177: No Sound / sound muting on nightly build

Rstation4, 2021-02-10:
Raspberry Pi 4 B 4gb
Lakka: Lakka-RPi4.arm-nightly-20210207-c96c51a

I have not tested all cores but so far the N64 Mupen does not have sound and of course when I went to test the other core it did not function or launch the game. Another core where I notice that sound was being muted or more like skipped is in the dreamcast core FLycast.  Desmune did not have any sound when lauching roms, I need further testing to see if Desmune 2015 has sound skipping/muting.

gouchi, 2021-02-21:
Uncomment/add `hdmi_drive=2` to `config.txt` does not help ?

[Source](http://www.lakka.tv/doc/Raspberry-Pi/#audio-not-working-on-some-tvs)

Rstation4, 2021-02-24:
I ended up wiping that sd card and reinstalling a newer nightly, I haven't fiddled to much on games but on RPG Maker there are instances of sound muting/ sound cutting off. I may have heard a tiny fraction of the sound being temporarily cut/muted on another core, but I will not be able to continue today or maybe tomorrow to test that out since I gave that unit away to introduce someone to Lakka and emulation. I will have to hear from that person if there was any additional problems.

Maybe unrelated but I was getting a error on Yume Nikki english (uboachan fan translation ver) where it will freeze everything and needed a forced powering off the RPI4b.

gouchi, 2021-02-28:
So at the end did you succeed to have sound working ? You have this sound issue for all core ?

Rstation4, 2021-03-01:
I did not get to fully test all the cores before giving away that Raspberry pi 4 setup, I gave it to a relative and they did mention to me later about N64 having no sound. I will ask them about it the next time I see them.

Rstation4, 2021-03-03:
I went ahead and used another RPi 4B 4gb Unit and used this nightly build **Lakka-RPi4.arm-nightly-20210227-e925215** on it. I noticed on Dreamcast, that the sound will cut off like a hiccup for a short bit. 

On **Desmune** it does not play sound which I tried on two separate roms. But for **Desmune 2015** it does play sound although not perfectly, because it sounds like you do get a sound hiccup once in a while, the second rom that was zipped did not play any sound. Will try with a few more roms later.

**Sega Saturn (Yabuse)**: I get no audio when I was trying out a single rom, a little bit of sound was working on the startup bios logo but that was all.  Rom tested- Keio Flying Squadron 2 (Europe)

N64 **Parrallel** does not work still, and **Mupen** does not have sound. Tested only two roms shortly with these two cores.

gouchi, 2021-03-04:
So you've got some sound not no sound...

Please check this [documentation](http://www.lakka.tv/doc/Timezone-settings/) for the timezone.

Rstation4, 2021-03-04:
In some instances there is a bit of sound being temporarily muted. I have more logs from another core.
[retroarch DC and Psx SM.log](https://github.com/libretro/Lakka-LibreELEC/files/6086619/retroarch.DC.and.Psx.SM.log)

I have also encountered in that particular psx rom that the menu screen input does not work or allow me to get past the menu, similar to "The Legend of Mana" rom. While testing a rom a psx rom after that one at a later time I was experiencing audio issues again.

Rstation4, 2021-03-12:
I just used the new nightly build and it still has the same persistent no sound/sound muting bug. Several cores still affected by this.
Lakka-RPi4.arm-nightly-20210312-2e60af4

Rstation4, 2021-03-15:
This is a very short 3 second clip of lagging and sound mutting/hiccups on the PPSSPP core.

https://user-images.githubusercontent.com/73009731/111112448-1623e700-8558-11eb-918a-4e7ae3f9eaa6.mp4

Since Ive been on this issue it varies by core and rom. Some roms will get to a particular part till you hear a sound issue. 

https://user-images.githubusercontent.com/73009731/111112950-ede8b800-8558-11eb-8a91-c605b3a7a41c.mp4

Rstation4, 2021-04-06:
Used the latest nightly build for RPI4 and there's still that audio issue. Only core I tested so far was Dreamcast's Flycast core.

dmrlawson, 2021-04-08:
I've found changing the audio driver to `alsathread` solves this for me. If it works for others I suggest we set that as default

Rstation4, 2021-05-30:
In the past few nightly builds before I moved on to the stable release I have not heard any audio issues. Should we considered this issue fixed and closed?

dmrlawson, 2021-05-30:
Sounds good to me

closed 2021-05-30 by dmrlawson