libretro/Lakka-LibreELEC #1178: Documentation updating/adding

Rstation4, 2021-02-10:
Raspberry Pi 4 B 4gb

I have recently got into MS-Dos and found that it was another frustrating and time consuming process of getting to work. Lack of current info that is applicable to a certain hardware and version. 

1. Why is there no documentation on Lakka of the difference in the two cores Dosbox and Dosbox-svn
2. What windows version do they actually support for our RPi4 arm Lakka version, some wiki's say that DosBox supports windows 3.x games but when I tried to get them to work they did not work on my platform and Lakka build.
3. DosBox-x  is that going to be added to the list of cores?  https://dosbox-x.com/

Retroarch documentation: https://docs.libretro.com/library/dosbox/ 
Lakka documentation: https://www.lakka.tv/doc/DOSBox/  

I don't know if Dosbox actually supports some windows 9 games but according to wikipedia it does.
 https://en.wikipedia.org/wiki/DOSBox

It would be nice for documentation to be updated so others can be saved from wasting their time and getting frustrated like I have been, Its not just with this system but other systems as well which I personally have experienced. Eg SNK Neo Geo CD

I just recently updated to a newer nightly build but I already discovered some sound issues with it.

Ntemis, 2021-02-10:
Documentation updates is something we don't have time for.
If interested please step up

gouchi, 2021-02-21:
The source documentation is the [wiki](https://github.com/libretro/Lakka-LibreELEC/wiki).

You can try to submit patch using 
`git clone https://github.com/libretro/Lakka-LibreELEC.wiki.git`

Thank you for your help.

Rstation4, 2021-03-15:
I see it's been a while since the wishlist has been updated as well. I do have a suggestion of [LiberaPay](https://liberapay.com/), I have heard of SubscribeStar as a alternative means of getting donations.

Ive been able to change the timezone via file manager and text editor on Linux as a alternative, only thing is how does one specifically change the time to an exact setting of Hours-Minutes-seconds instead of just the timezone?

closed 2021-07-26 by Ntemis