libretro/Lakka-LibreELEC #1181: Cannot switch between drivers

ribhutheincompetent, 2021-02-15:
Tried the new Doom 3 core, but stated that it needs OpenGL considering that the drivers are defaults to Vulkan. Alright, switch to Vulkan, and then restarted the shell. There's nothing but the void on the screen and the flash drive was in the constant state of 'thinking' for 40 minutes. Realized that that is not gonna work and thought the good old .cfg restart would do the trick but the /storage partition is somehow 'lockout' as I didn't have permissions to write on them.

Ntemis, 2021-02-15:
There is no vulkan support yet

ToKe79, 2021-02-15:
and the partition is probably full - due to core-dumps written to it - check `/storage/.cache/cores` folder (iirc) and remove it if you see a lot of "core-dump" files there.

gouchi, 2021-04-21:
@ribhutheincompetent is it still an issue with [latest nightly](http://nightly.builds.lakka.tv/latest/) ?

ToKe79, 2021-04-21:
Also please note following - if you mount the Lakka storage partition under other Linux OS, the files/folders on this partition are owned by uid/gid=0 / root (as Lakka is running in kind of a single user mode with root privileges), so you need to either switch to root to make modifications to files/folders on the storage (LAKKA-DISK) partition or take ownership of those files/folders you want to modify.

closed 2021-07-13 by ToKe79