libretro/Lakka-LibreELEC #1184: Segmentation Fault when browsing playlists with thumbnails - rpi2/3

jdalmanza, 2021-02-21:
lakka rpi2.arm nightly (tested most recently on Lakka-RPi2.arm-nightly-20210213-8076be1.img.gz)

When browsing playlists with thumbnails, after a while retroarch crashes with a segmentation fault

I tested this behavior on x86-64 and ofroid go advance (using the same playlists and thumbnails) and the behavior only seems to exist on the rpi image.

[INFO] RetroArch 1.9.0 (Git 5e551dd92b)
[INFO] === Build =======================================
[INFO] Capabilities:  NEON VFPv3 VFPv4
[INFO] Built: Feb 12 2021
[INFO] Version: 1.9.0
[INFO] Git: 5e551dd92b
[INFO] =================================================
[INFO] [Environ]: SET_PIXEL_FORMAT: RGB565.
[INFO] [Overrides]: Redirecting save file to "/storage/savefiles/.srm".
[INFO] [Overrides]: Redirecting save state to "/storage/savestates/.state".
[INFO] Version of libretro API: 1
[INFO] Compiled against API: 1
[INFO] [Audio]: Set audio input rate to: 48000.00 Hz.
[INFO] [Video]: Video @ fullscreen
[INFO] [Video]: Starting threaded video driver ...
[INFO] [DRM]: Found 2 connectors.
[INFO] [DRM]: Connector 0 connected: yes
[INFO] [DRM]: Connector 0 has 31 modes.
[INFO] [DRM]: Connector 0 assigned to monitor index: #1.
[INFO] [DRM]: Connector 1 connected: no
[INFO] [DRM]: Connector 1 has 1 modes.
[INFO] [DRM]: Mode 0: (1360x768) 1360 x 768, 60 Hz
[INFO] [DRM]: Mode 1: (1920x1080) 1920 x 1080, 60 Hz
[INFO] [DRM]: Mode 2: (1920x1080) 1920 x 1080, 60 Hz
[INFO] [DRM]: Mode 3: (1920x1080i) 1920 x 1080, 60 Hz
[INFO] [DRM]: Mode 4: (1920x1080i) 1920 x 1080, 60 Hz
[INFO] [DRM]: Mode 5: (1920x1080) 1920 x 1080, 50 Hz
[INFO] [DRM]: Mode 6: (1920x1080i) 1920 x 1080, 50 Hz
[INFO] [DRM]: Mode 7: (1920x1080) 1920 x 1080, 24 Hz
[INFO] [DRM]: Mode 8: (1920x1080) 1920 x 1080, 24 Hz
[INFO] [DRM]: Mode 9: (1280x768) 1280 x 768, 60 Hz
[INFO] [DRM]: Mode 10: (1280x720) 1280 x 720, 60 Hz
[INFO] [DRM]: Mode 11: (1280x720) 1280 x 720, 60 Hz
[INFO] [DRM]: Mode 12: (1280x720) 1280 x 720, 50 Hz
[INFO] [DRM]: Mode 13: (1024x768) 1024 x 768, 75 Hz
[INFO] [DRM]: Mode 14: (1024x768) 1024 x 768, 70 Hz
[INFO] [DRM]: Mode 15: (1024x768) 1024 x 768, 60 Hz
[INFO] [DRM]: Mode 16: (800x600) 800 x 600, 75 Hz
[INFO] [DRM]: Mode 17: (800x600) 800 x 600, 72 Hz
[INFO] [DRM]: Mode 18: (800x600) 800 x 600, 60 Hz
[INFO] [DRM]: Mode 19: (720x576) 720 x 576, 50 Hz
[INFO] [DRM]: Mode 20: (720x576i) 720 x 576, 50 Hz
[INFO] [DRM]: Mode 21: (720x480) 720 x 480, 60 Hz
[INFO] [DRM]: Mode 22: (720x480) 720 x 480, 60 Hz
[INFO] [DRM]: Mode 23: (720x480) 720 x 480, 60 Hz
[INFO] [DRM]: Mode 24: (720x480i) 720 x 480, 60 Hz
[INFO] [DRM]: Mode 25: (720x480i) 720 x 480, 60 Hz
[INFO] [DRM]: Mode 26: (640x480) 640 x 480, 75 Hz
[INFO] [DRM]: Mode 27: (640x480) 640 x 480, 73 Hz
[INFO] [DRM]: Mode 28: (640x480) 640 x 480, 60 Hz
[INFO] [DRM]: Mode 29: (640x480) 640 x 480, 60 Hz
[INFO] [DRM]: Mode 30: (720x400) 720 x 400, 70 Hz
libGL: Can't open configuration file /etc/drirc: No such file or directory.
libGL: Can't open configuration file /storage/.drirc: No such file or directory.
libGL: Can't open configuration file /etc/drirc: No such file or directory.
libGL: Can't open configuration file /storage/.drirc: No such file or directory.
libGL: Can't open configuration file /etc/drirc: No such file or directory.
libGL: Can't open configuration file /storage/.drirc: No such file or directory.
libGL: Can't open configuration file /etc/drirc: No such file or directory.
libGL: Can't open configuration file /storage/.drirc: No such file or directory.
[INFO] [GL]: Found GL context: kms
[INFO] [GL]: Detecting screen resolution 1360x768.
[INFO] [EGL] Found EGL_EXT_platform_base, trying eglGetPlatformDisplayEXT
[INFO] [EGL]: EGL version: 1.4
[... 309 lines cut ...]
function is no-op
function is no-op
function is no-op
function is no-op
function is no-op
function is no-op
function is no-op
function is no-op
function is no-op
function is no-op
function is no-op
function is no-op
function is no-op
function is no-op
function is no-op
function is no-op
function is no-op
function is no-op
Segmentation fault

for each 'function is no-op' in that output matches with controller input browsing the playlist until it crashes with the seg fault

Occasionally I see a full stack trace when this happens, I'll try to make it happen again and see if I can capture it.

Please let me know what other logs or data you might need.

jdalmanza, 2021-02-21:
tested on devel-20210220012040-424b1d8, same behavior
[log1.txt](https://github.com/libretro/Lakka-LibreELEC/files/6018533/log1.txt)

###########################################
# Lakka - The DIY retro emulation console #
# ...... visit http://www.lakka.tv ...... #
###########################################

Lakka (official): devel-20210220012040-424b1d8 (RPi2.arm)

gouchi, 2021-02-21:
Do you get the issue if you disable menu shader pipeline by going to  `Settings tab > User Interface > Appearance > Menu Shader Pipeline > Off` ?

jdalmanza, 2021-02-21:
> Do you get the issue if you disable menu shader pipeline by going to `Settings tab > User Interface > Appearance > Menu Shader Pipeline > Off` ?

Yes. Just now

Ntemis, 2021-02-21:
you should open an issue on retroarch issues tracker

jdalmanza, 2021-02-21:
> you should open an issue on retroarch issues tracker

ok, it's hard to tell if it was a lakka bug or retroarch bug, will do so.

Ntemis, 2021-06-06:
Can we close this one down?

jdalmanza, 2021-06-07:
Depends, If the answer is "this is a limitation of the pi3 because of limited RAM", then yes, you can close it down. I have not checked on 3.1 but last time I tried on one of the recent nightlies the behavior still existed, so I will try a more recent build (3.1 or more recent nightly) and post results here

jdalmanza, 2021-06-07:
behavior still exists on 3.1 build for rpi2.arm. Could be that this issue is addressed in an aarch64 build for pi3, hard to tell.
Let me know if you want to see additional logs.

jdalmanza, 2021-06-08:
FYI, I tried the same test on Lakka-RPi3.aarch64-3.0-devel-20210607220609-1e63d65.img.gz (thanks @dmrlawson ) and while it did not crash RA this time, i did get it to lock up once. 
I tried 2 playlists with 25 and 35 roms and full thumbnails (left and right) and did not seg fault this time. I will be testing on much larger romsets in the future to see if i can make it repeat the behavior from the 32bit build, but so far this looks promising

jdalmanza, 2021-06-12:
confirmed this still does occur on the aarch64 version (nightly build from 10-June) it does take longer to happen, then the 32-bit builds. Got it to crash RA using the full neogeo mvs romset with thumbs. i'll grab logs and attach shortly

closed 2021-07-23 by Ntemis