libretro/Lakka-LibreELEC #1185: Gamecon module is not enabled in the latest Lakka x86_64 version(Lakka (official): devel-20200902003240-291d416 (Generic.x86_64))

BDJoe360, 2021-02-23:
This issue tracker is NOT for help or support. Please use the Lakka forum (https://forums.libretro.com/c/libretro/lakka-tv-general) or IRC @ #lakka on freenode.net for those kinds of posts.

BEFORE you report a bug make sure you have tried the latest nightly version of Lakka at http://le.builds.lakka.tv/. Your bug might be already fixed.

If you are sure that it's a bug in Lakka, please answer the following questions:
- Which version of Lakka are you using?
- What system hardware are you using?
- What did you do?
- What did you expect to happen?
- What happened instead?

It is important to provide logs for debugging as part of your report. Instructions for generating logs can be found in the Troubleshooting Lakka documentation at http://www.lakka.tv/doc/Troubleshooting-Lakka/

Please delete this text before posting.

gouchi, 2021-02-28:
Hi,

Can you make a test with this  [image](https://we.tl/t-zMeduF3Dno) ?

Thank you.

Kind regards,

BDJoe360, 2021-03-03:
Hi, Gouchi
Thank you for your letter and test file.
I tested the file in an upgrade way,The good news is:
After checking with putty, you can see that gamecon module has taken effect：

But now there is another, more serious problem
The system just passed this screen when it started：

And then it's stuck in this picture:

And the old file will not restore the system to normal
The hash value and log of your file are attached here. I hope you can solve this problem

Thank you.

Kind regards！

 
Hi,
 
Can you make a test with this  image ?

gouchi, 2021-03-04:
I am glad gamcon module is working.

It seems the screenshots are missing.

commit c17fc1d0 (2021-07-14): generic 64 bit: add back gamecon support

fixes #1185
closed 2021-07-14 by ToKe79