libretro/Lakka-LibreELEC #1187: Build fails for RK3328

dtorbo, 2021-02-26:
- Which version of Lakka are you using?
Following the docs, I cloned Lakka to build:
`$ git clone https://github.com/libretro/Lakka-LibreELEC`

- What system hardware are you using?
Linux, amd64

- What did you do?
I tried to compile for RK3328.
`DISTRO=Lakka PROJECT=Rockchip ARCH=arm DEVICE=RK3328 make image`

- What did you expect to happen?
The build completes.

- What happened instead?
The build failed at Tic80.

Presumably, this will fail for anyone cloning the repo, right now.

```
[306/327] [DONE] install theodore
BUILD      tic80 (target)
    TOOLCHAIN      cmake
Executing (target): cmake -GNinja -DCMAKE_EXPORT_COMPILE_COMMANDS=ON -DCMAKE_TOOLCHAIN_FILE=/opt/lakka/Lakka-LibreELEC/build.Lakka-RK3328.arm-3.0-devel/toolchain/etc/cmake-armv8a-libreelec-linux-gnueabi.conf -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_BUILD_TYPE=MinSizeRel -DBUILD_PLAYER=OFF -DBUILD_SDL=OFF -DBUILD_SOKOL=OFF -DBUILD_DEMO_CARTS=OFF -DBUILD_LIBRETRO=ON /opt/lakka/Lakka-LibreELEC/build.Lakka-RK3328.arm-3.0-devel/tic80-8bfb0de
On branch master
Your branch is behind 'origin/master' by 245 commits, and can be fast-forwarded.
  (use "git pull" to update your local branch)

Untracked files:
  (use "git add <file>..." to include in what will be committed)

        .armv8a-libreelec-linux-gnueabi/
        .libreelec-package
        .libreelec-unpack

nothing added to commit but untracked files present (use "git add" to track)
Building for target : Linux
PROJECT_VERSION: 0.80.1281-dev
CMake Error at CMakeLists.txt:530 (add_subdirectory):
  The source directory

    /opt/lakka/Lakka-LibreELEC/build.Lakka-RK3328.arm-3.0-devel/tic80-8bfb0de/vendor/curl

  does not contain a CMakeLists.txt file.

CMake Error at CMakeLists.txt:573 (add_subdirectory):
  The source directory

    /opt/lakka/Lakka-LibreELEC/build.Lakka-RK3328.arm-3.0-devel/tic80-8bfb0de/vendor/zip

  does not contain a CMakeLists.txt file.

-- Configuring incomplete, errors occurred!
See also "/opt/lakka/Lakka-LibreELEC/build.Lakka-RK3328.arm-3.0-devel/tic80-8bfb0de/.armv8a-libreelec-linux-gnueabi/CMakeFiles/CMakeOutput.log".
FAILURE: scripts/install tic80 has failed!
[307/327] [FAIL] install tic80

The following logs for this failure are available:
  stdout: /opt/lakka/Lakka-LibreELEC/build.Lakka-RK3328.arm-3.0-devel/.threads/logs/310/stdout
  stderr: /opt/lakka/Lakka-LibreELEC/build.Lakka-RK3328.arm-3.0-devel/.threads/logs/310/stderr

parallel: This job failed:
package_worker 3 310 327 'install tic80'
Parallel build failure - see log for details. Time of failure: Fri Feb 26 00:31:08 PST 2021
make: *** [Makefile:12: image] Error 1
```

gouchi, 2021-02-28:
Hi,

[Nightly builds ](http://nightly.builds.lakka.tv/) are compiling fine.

As you [did](https://forums.libretro.com/t/tic80-build-fails-for-target-rk3328/32369/8), you should clean the source directory of tic80.

Thank you.

asdf288, 2021-03-01:
@gouchi, could it be that the nightly builds have cached the core build? The build fails if you try to build it from scratch, I've had this happen too. As mentioned in the libretro thread, the tic80 repository contains submodules, and some of the submodules have since been moved/deleted. If you don't have these sources cached anymore, the build will fail. To fix this, simply update the core to a recent git version.

gouchi, 2021-03-01:
Yes there is some cache indeed.

The libretro cores will be updated anyway with this PR #1122

ToKe79, 2021-03-03:
```
./get_libretro_tarball.sh tic80
```
will download cached package from the nightly server.

closed 2021-04-11 by davidgfnet