libretro/Lakka-LibreELEC #1188: Lakka not enabling USB on CM4-IO Board

jmacato, 2021-02-28:
> Which version of Lakka are you using?

Nightly (Lakka-RPi4.aarch64-nightly-20210227-e925215)

> What system hardware are you using?

Raspberry Pi Compute Module 4 (with eMMC) + CM4 IO Board.

> What did you do?

So i flashed the nightly build into the CM4 and also added the dtoverlay option needed to activate its usb hub. But whatever i tried, it still didnt enable the usb hub for some reason.

> What did you expect to happen?

Being able to use the USB.

> What happened instead?

Didnt work.

SethBarberee, 2021-05-22:
Is this still occurring for you? Firmware has been updated since the opening of this.

othermod, 2021-05-26:
I just tried with the latest nightly build. USB is still not working, even with the dtoverlay=dwc2,dr_mode=host added to config.txt

jmacato, 2021-05-26:
@SethBarberee i can confirm @othermod 's findings.

yikesmeguts, 2021-05-26:
Although I do not have the CM4 IO board, I do have a CM4 Lite and a GPiMate+  CM4.  Using the Pi4.arm Lakka build, and applying the GPiCase-specific settings, Lakka will boot on the GPiMate+ CM4, but the system will not respond to any input from the dpad.  Having used the same GPiCase-specific settings to configure RetroPie, I can confirm the hardware does work properly when dtoverlay=dwc2,dr_mode=host is defined in config.txt for RetroPie, but not for Lakka on the CM4. I do realize that Lakka and RetroPie are vastly different systems, but the above fix doesn't seem to apply in this case.

On an unrelated note, I also have a GPiMate+ CM3 and Lakka runs beautifully on the CM3.  Once the screen orientation issue has been fixed, it will be the perfect OS for the GPiCase, though I would prefer to use the CM4.

closed 2021-07-23 by Ntemis

jmacato, 2021-07-23:
@Ntemis may i ask why is this closed? The issue isnt fixed yet i think

Ntemis, 2021-07-23:
Sure, we don't have the device so I do not think a blind fix will be durable in the long run

dmrlawson, 2021-09-16:
Hi, I think this should work now in the latest nightlies if you enable it in distroconfig.txt