libretro/Lakka-LibreELEC #1189: RPG Maker sound and error screen

Rstation4, 2021-03-03:
RPI 4 Model B 4gb
Lakka Version: Lakka-RPi4.arm-nightly-20210227-e925215

With previous nightly builds I had sound muting/ hiccups and in some games that ran fine on my PC build ended up giving me a red screen of death on the RPi platform which then froze everything with no shutdown being possible without turning off the power.

[image]

gouchi, 2021-03-07:
It is maybe related to this [issue](https://github.com/EasyRPG/Player/issues/2470) upstream.

Ghabry, 2021-03-07:
My guess here is that liblcf somehow links against the wrong ICUdata library. Considering that you use the system libraries for compiling the cores that error doesn't make much sense.

@gouchi  is it possible to get the CMakeCache.txt of liblcf from a buildbot run? Otherwise I can't investigate this.

gouchi, 2021-03-07:
- I tested some [games from itch.io](https://itch.io/games/made-with-easyrpg) on x86_64 and arm and they are working.

- Then I tried Yume Nikki and I reproduce your issue on x86_64 and arm.

- Strangely enough I don't reproduce the issue using Linux x86_64 core  from buildbot on my PC.

@Ghabry [Here](https://github.com/libretro/Lakka-LibreELEC/blob/master/packages/devel/liblcf/package.mk#L20) is the version we are using on Lakka for liblcf.

It seems a new patch was added [liblcf-02-add-icu-compile-defs.patch](https://github.com/libretro/Lakka-LibreELEC/blob/master/packages/devel/liblcf/patches/liblcf-02-add-icu-compile-defs.patch) which is maybe the issue ?

Ghabry, 2021-03-07:
It uses "x86_64-libreelec-linux-gnu/sysroot/usr/lib/libicudata.so" which looks fine to me. Can you tell me the filesize and the output of ``objdump -TC x86_64-libreelec-linux-gnu/sysroot/usr/lib/libicudata.so" ?

gouchi, 2021-03-07:
@Ghabry  here is the ouput of objdump

```
DYNAMIC SYMBOL TABLE:
0000000000002000 g    D  .dynamic       0000000000000000 _end
0000000000002000 g    D  .dynamic       0000000000000000 _edata
0000000000002000 g    D  .dynamic       0000000000000000 __bss_start
0000000000000280 g    DO .rodata        0000000000000038 icudt61_dat
```

Ghabry, 2021-03-07:
hm that looks fine to me so I'm unsure where the problem is :/. Have to investigate this.

Also that it only fails with Japanese games is strange. When the datafile lacks all data any game will fail...

Ghabry, 2021-03-09:
@gouchi 
Is ICU built from source via this mk file here?

https://github.com/libretro/Lakka-LibreELEC/blob/d8b096e05e1440c3730971f9a120a0b88916a644/packages/addons/addon-depends/chrome-depends/icu/package.mk

Ghabry, 2021-03-12:
@gouchi 
One last thing you have to investigate for me. As I just fixed mingw build this gives me another hint about the problem:
Can you tell me the filesize of libicudata.so and libicudata.a?

gouchi, 2021-03-12:
@Ghabry 

libicudata.so is 18
libicudata.a is 1.3K

Ghabry, 2021-03-14:
Okay this is the reason. The data is not included 🙄. Will send you a patch later that should fix this...

Ghabry, 2021-03-14:
@gouchi 

I have no idea where the problem in the ICU buildsystem is. Somehow in some cases when crosscompiling ICU picks the stub datafile and not the normal datafile. We have this problem with mingw cross and emscripten but with no other platform. Is the first time I see this issue when crosscompiling a Linux.

Guess you will have to experiment a bit until it compiles.

You can invoke the following code (after building ICU) which will force a datafile build:

```
# ICU: Force datafile build
export PKGDATA_OPTS="-w -v -O $PWD/icu/source/config/pkgdata.inc"

cd icu/source/data
make clean
make
```

Then copy the datafile to the system path (make install wont work, this still copies the stub...)

```
cp icu/source/lib/libicudt.* /usr/lib/
```

asdf288, 2021-03-16:
After some testing, I found that the data library (libicudata.a) needs a tool from the icu library itself, "bin/pkgdata", to be created properly. This tool doesn't exist because "--disable-tools" is set in the [package build file](https://github.com/libretro/Lakka-LibreELEC/blob/d8b096e05e1440c3730971f9a120a0b88916a644/packages/addons/addon-depends/chrome-depends/icu/package.mk). Removing this line will allow the pkgdata program to be created and after that, the static library (libicudata.a) is 27 MB instead of 1 KB.

But apart from that, something is strange here, because I can not reproduce this error. On my build, I have the faulty libicudata.a with 1 KB size, but the games play fine for me, I can play Yume Nikki and other games without problems. I will post again here after more investigation.

Ghabry, 2021-03-16:
Ohhhh, you are right, I didn't see that ``PKG_CONFIGURE_OPTS_HOST`` also appends ``$PKG_ICU_OPTS``. Yeah this is obviously wrong. For the host build the tools must be enabled (everything else can stay disabled). This also explains why it didn't work automatically. I never had problems with the data file when crosscompiling to a linux.

In general there are two steps required when building ICU, not sure if LibreELEC does this correctly, I only see the with-cross-build step:

1. First step is a native build, you can disable basicly everything there _except_ the tools.
2. Second is a crosscompile, you must specify the path to your just compiled native ICU via ``--with-cross-build``.

asdf288, 2021-03-16:
Yes, please disregard what I said earlier, I edited my comment accordingly. I just noticed that I was using a modified build for myself where I had set to build the icu libraries as static for some reason. I switched it back to Lakka's original settings. I am now rebuilding the libraries and the image, let's see if the fixed libicudata library fixed it!

asdf288, 2021-03-16:
Now I remember why I set the icu libraries to static before. Using the shared libraries, I'm getting the following error:
`/usr/lib/liblcf.so.0: error while loading shared libraries: /usr/lib/libicudata.so.61: internal error`
And `ldd libicudata.so.61.1` says: `not a dynamic executable`. It seems as if the icu libraries are corrupted somehow. Any ideas?

Ghabry, 2021-03-16:
I have never seen that error before but I also never used custom built dynamic ICU library (always use the static one).

But I know from digging deep in the source years ago (had to figure out how to do a little -> big endian ICU cross build for the Wii, took me days...) that the shared library is not created by a compiler+linker but manually assembled by the pkgdata tool (I have really no idea why they are doing this...). Argument "-w" will force compile+link but there is no direct way to enforce this.

You could try the commands I linked above if they yield a shared library (they force usage of compile+link):

```
# ICU: Force datafile build
export PKGDATA_OPTS="-w -v -O $PWD/icu/source/config/pkgdata.inc"

cd icu/source/data
make clean
make
```

asdf288, 2021-03-16:
@Ghabry thanks for your suggestion,
I could not get the make command to complete, it fails because it's trying to include files from /usr/include instead of the sysroot folder.

Before digging further into this, just to clarify: The libicudata.so library is already being generated with a size of 28 MB, but seems to be faulty. Would these commands create a different one?

Ghabry, 2021-03-16:
Can you tell me how to setup a Lakka-LibreELEC-compatible buildroot so I can also try to figure this out?

The command creates a slightly different so-file because it uses the cross-compiler instead of glueing it together by emitting binary. But that it fails for you because it looks in the wrong folder is strange, never had this before.

Afaik static libicudata is not really standard. We do it for our upstream builds because this makes distribution easier. We also use a custom data file that is only 700 kbyte in size because we do not need most of the Unicode database.

asdf288, 2021-03-16:
Sorry, I am not familiar with buildroot. I could only help if you want to build Lakka. This is how to compile it:
https://www.lakka.tv/doc/Compiling-Lakka/
`DISTRO=Lakka PROJECT=Generic ARCH=x86_64 scripts/build liblcf`

I have now completed the make command you suggested, by manually including the missing directory in the pkgdata.inc file. And it really did something different, it took a very long time and output a new libicudata.so at the end which is slightly bigger than the old one. This one does not show an error when running ldd. Seems promising, I'm going to test it now!

asdf288, 2021-03-16:
Success! The manually created libicudata.so works! Thank you @Ghabry ! Would you mind explaining how to create the smaller custom icu data file that you mentioned above? That would be another great improvement to include in Lakka, because right now, it's quite a heavy library, both in compile time and file size.

Ghabry, 2021-03-16:
I was finally able to get the build root up and running. I have no idea why it doesn't build the libicudata.so by default though. Always just the stubdata...

Here is the script how we create the data file. At least newer ones. Unfortunately your 61.2 ICU is too old and lacks the new python filtering script (needs ICU 67) and I cannot find our old script anymore. But we archive all old data files: https://easyrpg.org/downloads/icudata/

To use a custom data file simply copy it into the ICU source directory (of the target, not of the host):
```
cp icudt*.dat $ICU_DIR/source/data/in
```

Also _important_: The data file is stripped for EasyRPG Player needs (we only need some encoding tables and normalisation). Other consumers are maybe unhappy with this very limited file.

The script:
```
cd icu/source

# Write icudata filter file
# ibm-943_P15A-2003   # Japanese with \ as backslash
# ibm-1363_P11B-1998  # Korean with \ as backslash
# windows-949-2000    # Korean ?
# ibm-5346_P100-1998  # Central Europe with Euro
# ibm-5347_P100-1998  # Cyrillic with Euro
# ibm-5348_P100-1997  # Occidental with Euro
# ibm-5349_P100-1998  # Greek with Euro
# ibm-9447_P100-2002  # Hebrew with Euro
# ibm-9448_X100-2005  # Arabic with Euro + 8 chars
# windows-874-2000    # Thai with Euro
# windows-936-2000    # Chinese simplified
# (depends on ibm-1386_P100-2001)
# windows-950-2000    # Chinese traditional
# (depends on ibm-1373_P100-2002)
# ibm-5350_P100-1998  # Turkish
# ibm-9449_P100-2002  # Baltic
# nfkc                # Normalization (NFKC)
# convrtrs            # Encoding alias names

cat <<EOF > icudata-filter.json
{
  "strategy": "additive",
  "featureFilters": {
    "conversion_mappings": {
      "whitelist": [
        "ibm-943_P15A-2003",
        "ibm-1363_P11B-1998",
        "windows-949-2000",
        "ibm-5346_P100-1998",
        "ibm-5347_P100-1998",
        "ibm-5348_P100-1997",
        "ibm-5349_P100-1998",
        "ibm-9447_P100-2002",
        "ibm-9448_X100-2005",
        "windows-874-2000",
        "windows-936-2000",
        "ibm-1386_P100-2001",
        "windows-950-2000",
        "ibm-1373_P100-2002",
        "ibm-5350_P100-1998",
        "ibm-9449_P100-2002"
      ]
    },
    "normalization": {
      "whitelist": [
        "nfkc"
      ]
    },
    "cnvalias": {
      "whitelist": [
        "convrtrs"
      ]
    }
  }
}
EOF

# uses own ICU tools to avoid chicken-egg problem when using the static GNU/Linux toolchain
ICU_DATA_FILTER_FILE=icudata-filter.json ./configure \
  --prefix=$WORKSPACE --disable-shared --enable-static --with-data-packaging=archive \
  --disable-dyload --disable-extras --disable-icuio --disable-layout --disable-tests --disable-samples

# cleanup unneeded data file content
rm data/in/icudt*.dat

make

# remove unneeded and display result
cd data/out
export NOW="`date "+%Y-%m-%d %H:%M:%S"`"
echo 'coll/ucadata.icu unames.icu *.res' > useless.txt
../../bin/icupkg -r useless.txt -C "EasyRPG $NOW" -l icudt${IVD}l.dat
../../bin/icupkg -tb icudt${IVD}l.dat icudt${IVD}b.dat
ls -alh *.dat
```

asdf288, 2021-03-16:
Maybe it's triggered somewhere in the icu build script by setting --enable-tools in the configure script? (if there are no tools, we can't build the library)?

Thank you for the script to build the minimal library. I don't see why we couldn't update the icu libraries as liblcf is the only part of Lakka that depends on it. If I remember though, newer icu versions didn't compile with Lakka, but this should be fixable.

Ghabry, 2021-03-16:
Upstream we use "--enable-tools" for the host build and "--disable-tools" for the crossbuild because the crossbuild does not need tools (they won't run anyway).
I have no idea why but the crossbuild does not invoke the "pkgdata" tool from the host. Normally it should invoke it and create the libicudata.so.
This could be a bug with ICU61 maybe?

asdf288, 2021-03-16:
For some reason you can't disable the tools for the target build, even if they're never used. Try to enable them and the icudata library should be built.

Ghabry, 2021-03-16:
Yeah this must be a regression. I tried with 59.2 (we use this one) and it creates the "libicudata.so" for the cross-build, when host has "--enable-tools" and cross has "--disable-tools".
So this is indeed a change in behaviour here!

This also explains why my mingw libretro core build does not generate a data file. It uses ICU61! -_-

asdf288, 2021-03-17:
Updating icu from 61.2 to 68.1 seems to fix both of the previous issues (target does not need --enable-tools anymore, resulting library is not corrupted anymore, no workaround needed).

@Ghabry , I have now tried the script for reducing the size that you posted above. Your example is not for a cross-compile setup, correct? I assume that for Lakka, these steps should only be needed for the target build? I have incorporated it into the build script and I end up with two .dat files of about 650 KB in size in data/out. Now I am a bit lost, what do I do with these files? Are these a replacement for the static/dynamic libicudt library? Where do I put them, are these runtime files or are they needed for compiling liblcf? 

Another question, would it possible to build just the tools for the host, skipping the libraries altogether as they are not needed?

Ghabry, 2021-03-17:
I think they changed something again, my script doesnt work anymore for 68.1, I try to fix it.

asdf288, 2021-03-17:
No, you don't have to, it worked fine. It successfully produced the smaller libraries. It just needed to be adapted to Lakka's build system to make it work with cross-compiling.

My idea was to build everything as static (the custom ICU and liblcf), because this would allow to build the EasyRPG core in a way that it can easily be updated later. Lakka users can update libretro cores, but not system libraries, so the cores should have as few dependencies as possible. When I'm able to finish it, I will create a pull request with the changes.

Ghabry, 2021-03-17:
> Your example is not for a cross-compile setup, correct? I assume that for Lakka, these steps should only be needed for the target build?

The dat generation stuff is for a host build. It basicly generates a dat-file and you can archive this dat file somewhere and then recycle it.

> At this step after running configure:

The data/in is in the "host-icu/source" folder.

and i forgot something you also need at the beginning:

```
wget https://github.com/unicode-org/icu/releases/download/release-68-1/icu4c-68_1-data.zip
tar xf icu4c-$IVU-src.tgz
unzip -o -q icu4c-$IVU-data.zip -d icu/source
```

> Now I am a bit lost, what do I do with these files?

You must copy the dat files into "source/data/in" of the _Cross_ build and then build as usual (shared library).

----

Btw here is the generated dat file:
[icudt68.zip](https://github.com/libretro/Lakka-LibreELEC/files/6156440/icudt68.zip)

Rstation4, 2021-03-18:
Using the latest nightly build I did another test run and Video record. Yes the build still has sound issues. at the 8 second to 9 second point is where you can hear a very short sound muting. 

[RPG Maker log 2.log](https://github.com/libretro/Lakka-LibreELEC/files/6160929/RPG.Maker.log.2.log)

Short Clip: The Rom was from RPGMaker.net 

https://user-images.githubusercontent.com/73009731/111565389-2b388a00-8793-11eb-8983-81c52718a468.mp4

PR #1221 (2021-03-23): Fix missing icudata library and runtime error
The previous icu package built an empty libicudata.so library and was missing a patch from upstream LibreELEC that fixes a runtime error on some platforms. This should fix #1189.
closed 2021-03-23 by Ntemis