libretro/Lakka-LibreELEC #1190: ScummVM game not adding or launching from .scummvm file

Rstation4, 2021-03-03:
Raspberry Pi 4 Model B 4gb
build: Lakka-RPi4.arm-nightly-20210227-e925215

One of the games "Day of the Tentacle" does not add when using directory scan and even though I added it via a manual scan and tried launching it via the created .scummvm file, it does not work or launch. I have however launched it via load content from one of the files in the folder and it works. Is this a database error or something because I had 2 other titles that were added to the playlist and launched.

RobLoach, 2021-03-03:
Could you share your .scummvm file? I'm seeing these entries...

```

game (
	name "Maniac Mansion: Day of the Tentacle (Floppy/DOS)[version a]"
	description "Maniac Mansion: Day of the Tentacle (Floppy/DOS)[version a]"
	code "tentacle"
	rom ( name "TENTACLE.000" size 7937 crc 37c17114 md5 acad97ab1c6fc2a5b2d98abf6db4a190 sha1 26292b98f4054f046b7dc8a7f52f89fd2857a942 )
)

game (
	name "Maniac Mansion: Day of the Tentacle (Floppy/DOS)[version b]"
	description "Maniac Mansion: Day of the Tentacle (Floppy/DOS)[version b]"
	code "tentacle-1"
	rom ( name "Maniac Mansion: Day of the Tentacle (Floppy/DOS)[version b].scummvm" size 10 crc 84b80509 )
)

game (
	name "Maniac Mansion: Day of the Tentacle (Floppy/DOS)[version b]"
	description "Maniac Mansion: Day of the Tentacle (Floppy/DOS)[version b]"
	code "tentacle-1 CR"
	rom ( name "Maniac Mansion: Day of the Tentacle (Floppy/DOS)[version b] CR.scummvm" size 11 crc d5eb9391 )
)

game (
	name "Maniac Mansion: Day of the Tentacle (Floppy/DOS)[version b]"
	description "Maniac Mansion: Day of the Tentacle (Floppy/DOS)[version b]"
	code "tentacle-1 CRLF"
	rom ( name "Maniac Mansion: Day of the Tentacle (Floppy/DOS)[version b] CRLF.scummvm" size 12 crc b50a4ed2 )
)

game (
	name "Maniac Mansion: Day of the Tentacle (Floppy/DOS)[version b]"
	description "Maniac Mansion: Day of the Tentacle (Floppy/DOS)[version b]"
	code "tentacle-1 LF"
	rom ( name "Maniac Mansion: Day of the Tentacle (Floppy/DOS)[version b] LF.scummvm" size 11 crc 4b8f0632 )
)

game (
	name "Maniac Mansion: Day of the Tentacle (Floppy/DOS)[v1.2]"
	description "Maniac Mansion: Day of the Tentacle (Floppy/DOS)[v1.2]"
	code "tentacle-3"
	rom ( name "TENTACLE.001" size 10049341 crc 5975b264 md5 415d17122ee8b8415c9b57c1062fb77d sha1 474c70c6788e119b20967302fdb185a9330981aa )
)
```
You could also try `tentacle-1` for the contents.

Rstation4, 2021-03-03:
@RobLoach I dont know where I'm supposed to view that which you are showing me, is that the database for it on Lakka OS? For the ones that worked some of them just have a empty file that's just has a game id name. For example mh2.scummvm, which did not need the echo line of code as listed in the lakka instructions. 

https://www.lakka.tv/doc/ScummVM/   I tried to get my games to work with using the directions here but it does not work for Maniac Mansion Day of the Tentacle (Floppy Dos version), Manhunter 1 (Floppy dos version), Island of Dr. Brain (Floppy DOS version).

For Day of the Tentacle (Floppy Dos version), scummvm file is tentacle.scummvm with the instructed codes from directions on Lakka documentation. echo tentacle >> tentacle.scummvm

 I tend to only use floppys since they seem to be easier to deal with, since I was messing with DOS box earlier and I just cant get some cd games to work and Windows 3.xx games. Would that be limited by the dosbox core provided. Any signs of Dosbox-x coming to LAKKA?

As for Day of the Tentacle, It does not say what version on the folder. I did a md5 check and its this 2723fea3dae0cb47768c424b145ae0e7     -From file TENTACLE.0

gouchi, 2021-03-07:
> I tend to only use floppys since they seem to be easier to deal with, since I was messing with DOS box earlier and I just cant get some cd games to work and Windows 3.xx games. Would that be limited by the dosbox core provided. Any signs of Dosbox-x coming to LAKKA?

As far as I know we don't have dosbox-x core, the new one is [dosbox-pure](https://github.com/libretro/dosbox-pure).

Unfortunately, the documentation is outdated as the database scan has evolved. 

If you have the files which match the database it will be added to the playlist.

I tried and it is working correctly.

RobLoach, 2021-03-07:
Before the ScummVM would index all files, and when there wasn't a unique file, it would depend on the .scummvm file. I've just pushed up a Pull Request which adds ALL .scummvm files: https://github.com/libretro/libretro-database/pull/1139

Rstation4, 2021-03-09:
Thanks, but seriously for all the work you guys do how come you don't have some kind of donations inbox like [LiberaPay](https://liberapay.com/). Or what about hardware donation? Ive been using Lakka for a while now and its a great asset to the emulation world. I wish Ludo OS can get there too.

ToKe79, 2021-06-20:
@Rstation4 can this be considered as solved? Thanks.

RobLoach, 2021-06-20:
The update to the database has been pushed forwards, so this should be good to close. Thanks for the ping, ToKe79!

closed 2021-06-21 by Ntemis