libretro/Lakka-LibreELEC #1191: Build fails for OdroidGoAdvance

temrix, 2021-03-03:
- Which version of Lakka are you using?
development version on master branch
- What system hardware are you using?
Debian Linux amd64 as guest OS in virtual machine (host OS also Debian amd64)
- What did you do?
Compile using `PROJECT=Rockchip DEVICE=OdroidGoAdvance ARCH=arm make image`.
- What did you expect to happen?
Compiles.
- What happened instead?
Linux kernel did not compile.
```
drivers/net/wireless/rtl8821cu/hal/phydm/halrf/rtl8821c/halrf_iqk_8821c.c: In function '_iqk_reload_iqk_setting_8821c':
drivers/net/wireless/rtl8821cu/hal/phydm/halrf/rtl8821c/halrf_iqk_8821c.c:579:49: warning: '~' on a boolean expression [-Wbool-operation]
error, forbidden warning:halrf_iqk_8821c.c:579
     odm_set_bb_reg(dm, iqk_apply[path], BIT(0), ~(iqk_info->iqk_fail_report[channel][path][idx]));
                                                 ^
drivers/net/wireless/rtl8821cu/hal/phydm/halrf/rtl8821c/halrf_iqk_8821c.c:579:49: note: did you mean to use logical not?
     odm_set_bb_reg(dm, iqk_apply[path], BIT(0), ~(iqk_info->iqk_fail_report[channel][path][idx]));
                                                 ^
                                                 !
drivers/net/wireless/rtl8821cu/hal/phydm/halrf/rtl8821c/halrf_iqk_8821c.c:581:50: warning: '~' on a boolean expression [-Wbool-operation]
error, forbidden warning:halrf_iqk_8821c.c:581
     odm_set_bb_reg(dm, iqk_apply[path], BIT(10), ~(iqk_info->iqk_fail_report[channel][path][idx]));
                                                  ^
drivers/net/wireless/rtl8821cu/hal/phydm/halrf/rtl8821c/halrf_iqk_8821c.c:581:50: note: did you mean to use logical not?
     odm_set_bb_reg(dm, iqk_apply[path], BIT(10), ~(iqk_info->iqk_fail_report[channel][path][idx]));
                                                  ^
                                                  !
```

Also, where does the toolchain pull the kernel from? Where can I set another kernel version?

temrix, 2021-03-03:
After excluding RTL8821CU from building, I get:
```
In file included from /home/temrix/Lakka-LibreELEC/build.Lakka-OdroidGoAdvance.arm-3.0-devel/toolchain/armv8a-libreelec-linux-gnueabi/sysroot/usr/include/string.h:494,
                 from fs/fs.c:7:
In function 'strncpy',
    inlined from 'fs__env_override' at fs/fs.c:191:2,
    inlined from 'fs__get_mountpoint' at fs/fs.c:197:6:
/home/temrix/Lakka-LibreELEC/build.Lakka-OdroidGoAdvance.arm-3.0-devel/toolchain/armv8a-libreelec-linux-gnueabi/sysroot/usr/include/bits/string_fortified.h:106:10: error: '__builtin_strncpy' specified bound 4096 equals destination size [-Werror=stringop-truncation]
  106 |   return __builtin___strncpy_chk (__dest, __src, __len, __bos (__dest));
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
  CC       util/annotate.o
  CC       fs/tracing_path.o
  CC       event-plugin.o
  CC       trace-seq.o
  CC       bpf.o
  LD       fs/libapi-in.o
```

davidgfnet, 2021-03-20:
This was fixed

closed 2021-03-20 by davidgfnet