libretro/Lakka-LibreELEC #1198: Graphical system glitch on PPSSPP core (Raspberry Pi 4)

Rstation4, 2021-03-09:
Raspberry pi 4 B 4gb
Lakka Nightly rpi4 arm feb-27 2021

This rom was getting this graphical error. I also included a performance counter log. I have the logs set to debug and timestamp so not all is lost upon freezing and I can copy and paste them to my pc. Does the performance counter logs help out or are they kinda worthless to post?

[image]

PPSSPP Log:
[PPSSPP run.log](https://github.com/libretro/Lakka-LibreELEC/files/6111858/PPSSPP.run.log)

Performance Log: 
[Performance Counter log.log](https://github.com/libretro/Lakka-LibreELEC/files/6111861/Performance.Counter.log.log)

Ntemis, 2021-06-06:
Please test 3.1 release

ToKe79, 2021-07-15:
To me this looks like issue with ppsspp-assets, which should be already included in the image, but adding them to `/storage/system` may cause an issue. Please check if you have the ppsspp-assets (iirc folder name `PPSSPP`) in `/storage/system` and if so, remove/rename the folder, reboot and try again. If the issue persists, please comment/re-open. Thank you!

closed 2021-07-15 by ToKe79