libretro/Lakka-LibreELEC #1205: Wifi issues with BCM43224 (2.3.2 & nightly)

IzaacJ, 2021-03-11:
- Which version of Lakka are you using?
   Generic.x86_64-2.3.2
- What system hardware are you using?
   Dell Latitude E6410
- What did you do?
   Try to connect to wifi
- What did you expect to happen?
   Being able to scan and see available networks
- What happened instead?
   Garbled name in the wifi list, and unable to use connmanctl to connect.

Been trying to get this up and running but using the UI does not work, and using CLI through wired SSH session to setup wifi doesn't work either.

`connmanctl> enable wifi` generates `Error wifi: Method "SetProperty" with signature "sv" on interface "net.connman.Technology" doesn't exist`.

`lspci -nnk` gives me the output below, but at a first glance (02:00.0) it seems to detect the wifi chipset (BCM43224) and use the `bcma-pci-bridge` driver.

The nightly for today seems to have failed, but tried the nightly (live) from yesterday, again with no working wifi, but this time `connmanctl> enable wifi` gives me `wifi is not available`. `lspci -nnk` does, however, show the same drivers for the wifi chipset.

`lspci -nnk` From 2.3.2: https://pastebin.com/tCFyet5k

`dmesg` from 2.3.2: https://pastebin.com/TgvFbXL1

`lspci -nnk` from yesterdays nightly: https://pastebin.com/khjUjPa7

`dmesg` from nightly: https://pastebin.com/jL07pqZm

clysel, 2021-04-05:
I have the same issue, garbarge SSID after running wifi discovery.

But the i386 image worked for my setup 

Hardware name: Apple Inc. Macmini4,1/Mac-F2208EC8, BIOS     MM41.88Z.0042.B03.1111072100 11/07/11

03:00.0 Network controller [0280]: Broadcom Limited BCM43224 802.11a/b/g/n [14e4:4353] (rev 01)
        Subsystem: Apple Inc. BCM43224 802.11a/b/g/n (AirPort Extreme) [106b:0093]
        Kernel driver in use: bcma-pci-bridge
        Kernel modules: bcma, wl

by-gnome, 2021-05-27:
@IzaacJ Please try [latest nightly image](https://nightly.builds.lakka.tv/latest/Generic.x86_64/).

closed 2021-07-23 by Ntemis