libretro/Lakka-LibreELEC #1207: PC-98 (Neko Project II) Core game launches then exits

Rstation4, 2021-03-12:
Raspberry Pi 4 B 4gb
Lakka version: Lakka-RPi4.arm-nightly-20210310-6c07e38

Tested 5 of the freeware Touhou games made for **PC-98** (https://moriyashrine.org/files/category/36-retro-era/)
I tried all of them with the default settings and even started trying them with extended memory.

Logs: 
[PC-98 Touhou.log](https://github.com/libretro/Lakka-LibreELEC/files/6126699/PC-98.Touhou.log)

It will launch play the studio opening credit then default to a black screen
[image]

Rstation4, 2021-03-18:
Another test run
[retroarch pc98.log](https://github.com/libretro/Lakka-LibreELEC/files/6160706/retroarch.pc98.log)

https://user-images.githubusercontent.com/73009731/111560281-17d4f100-878a-11eb-9457-ed9754e8290a.mp4

Rstation4, 2021-05-26:
I thought I should revisit this and try again with a later stable build.
Lakka Version:  Lakka-RPi4.aarch64-3.0

So I've tested Touhou games 1,2 and 3. I know  1 and 2 work despite not working here on Lakka OS's  PC-98 core because I was able to run them on Retropie.

Logs:
[retroarch.log](https://github.com/libretro/Lakka-LibreELEC/files/6547676/retroarch.log)

Rstation4, 2021-07-09:
Update: Lakka 3.2 aarch64 stable build
Raspberry Pi 4 4gb RAM

It still gets that error. It has to be limited to Lakka's versions because this does not happen with current Retroarch 1.9.6 with the same core and bios files used. Even Retropies PC-98 neko project core worked with that  Touhou game "TH01 - Highly Responsive to Prayers.hdi"

gouchi, 2021-07-09:
Can you make a test with this [image](https://nightly.builds.lakka.tv/members/gouchi/1207/Lakka-RPi4.arm-3.0-devel-20210709212858-4f6ffc7.img.gz) as I added [np2kai core](https://docs.libretro.com/library/neko_project_ii_kai) ? 

The one we are currently provided is meowpc98.

Thank you.

Rstation4, 2021-07-10:
I have tested that image you provided and the 2 cores on that image on a Raspberry Pi 4 2gb version. Neko Project 2 core failed at that point again. The other core Neko Project II Kai worked. 

logs:
[retroarch.log](https://github.com/libretro/Lakka-LibreELEC/files/6796059/retroarch.log)

Ntemis, 2021-07-28:
As far as i can see arm64 is not supported by the core maybe we need to make it arm only this one

gouchi, 2022-03-02:
Closing this issue as OP is now ghost and workaround has been provided.

closed 2022-03-02 by gouchi