libretro/Lakka-LibreELEC #1212: Background image display corruption RPi 4

Rstation4, 2021-03-15:
Raspberry pi 4 B 4gb
Lakka Nightly build:

I've only tested this out on nightly builds, I do see a pattern on more complex wallpapers are showing pixels or not displaying the image right. 

Default image
[image]

Screenshot
[image]

ToKe79, 2021-03-23:
The original image is 1440x1080 so the corruption may come from resizing the image. Have you tried using the image with regular RetroArch (not Lakka), if the corruption appears as well?

Rstation4, 2021-03-23:
Ill try that later today, I currently dont have Retroarch installed on my Linux build on my PC.

DevilBlackDeath, 2021-04-01:
Had a similar issue. Have you by chance tried converting your image to png (if it isn't already this format), personally it helped !

Rstation4, 2021-05-24:
Raspberry Pi 4 (2gb)
RPi4 -arm-devel-20210520 
Before some of them that were converted did have issues as well.  I retested with 2 images that were jpeg  and converted them via Pix to .png at level 9 compression.

Jpeg Version (Original version) 1920x1080
[image]

Png converted version (Level 9 Compression) 1920x1080
[image]

dmrlawson, 2021-05-24:
Is it possible this image is relying on transparency (which jpeg does not support)?

DevilBlackDeath, 2021-05-25:
@dmrlawson Personally I tested it with around 10 wallpapers that were originally jpeg and had no issues at all on my desktop computer and had the same issue. Converting them to png solved the issue as well. There's something happening with jpeg's display code but I wouldn't be able to say what (though it's clearly around "borders" so probably something related to jpeg's compression).

dmrlawson, 2021-05-25:
> @dmrlawson Personally I tested it with around 10 wallpapers that were originally jpeg and had no issues at all on my desktop computer and had the same issue. Converting them to png solved the issue [...]

Ok, do you have a link to one of the images? On your desktop pc were you using Lakka or just retroarch (on windows? Linux?)

DevilBlackDeath, 2021-05-25:
One of these was that one : https://i.redd.it/g71062xwmsz51.jpg
Also, by working on my computer, I meant as images and normal wallpaper, not Retroarch's wallpaper. Though I did just test on Retroarch and it worked just fine.

dmrlawson, 2021-05-25:
OK, I can reproduce. Will investigate

Rstation4, 2021-08-04:
On Lakka 3.3 aarch64 stable version for the RPi 4 I still get that image issue. That image used as a background is a jpg file.

Screenshot:
[image]

dmrlawson, 2021-08-04:
Really not sure where to go with this.

@Ntemis would you mind trying this on a different device? I can reproduce in Lakka on rpi4 it's fine on Retroarch on my laptop. I suspect it's to do with how Retroarch decodes jpegs but I'm clueless as to whether it might be using the graphics driver for that.

Rstation4, 2021-10-31:
As for the new Lakka OS aarch64 3.5.2 Stable branch jpeg background images still show up incorrectly. (tested on a Full HD tv, with a Raspberry Pi 4 model B 4GB Ram hardware).

closed 2022-06-10 by Ntemis