libretro/Lakka-LibreELEC #1215: Lakka does not use nearest neighbor scaling on the UI when using integers of desktop resolution.

bagelbyheart, 2021-03-16:
I am attempting to tune lakka performance on an RPI3 and noticed that the system is not quite able to maintain a clean 60 FPS at 1080p, but does just fine at a half resolution of 960x540 (set in GUI, not hdmi mode). However the display becomes blurry and is clearly being bilinearly filtered.

The same issue occurs if you try to scale a display using xrandr with default settings, but can be worked around with a very simple fork of that. Of course since lakka doesn’t seem to use xorg or to have xrandr, I can’t really try that here, but it seems like the solution would be similarly applicable.

Some details about the xrandr adjustment are here: https://forums.linuxmint.com/viewtopic.php?t=159064

I know this feature will be really handy for me, and similar issues have come up in relation to using lakka with the rpi4 on 4k displays as well, for example, having this option would have also resolved the issue #871

SethBarberee, 2021-03-16:
Lakka on the RPi uses the KMS driver and doesn't use Xorg so if you want to mess with different resolutions, you'll want to mess with `config.txt`. Can try setting a different resolution vs Lakka picking one from EDID of the screen.

https://www.raspberrypi.org/documentation/configuration/config-txt/video.md

bagelbyheart, 2021-03-16:
@SethBarberee I'm referring to the GUI resolution and scaling, not the actual output resolution. So in this case, the `tvservice -s` output shows full 1080, but the GUI has been set to 960x540, and the filtering then happens when scaling that to the hdmi output. The hint to look at KMS seems like a start, but I still am having some trouble wrapping my head around it.

Based on this document: https://www.kernel.org/doc/html/latest/gpu/drm-kms.html

It seems like the issue has to do with the "DRM CRTC" but it's unclear if that's something set by the person making the video driver, or the person utilizing one.

bagelbyheart, 2021-03-16:
@SethBarberee this is also muddied a bit by some xorg stuff in debian which indicates that it can use the KMS driver to setup screens: https://xorg-team.pages.debian.net/xorg/howto/use-xrandr.html