libretro/Lakka-LibreELEC #1216: Fullscreen X and Y resolution can only be adjusted in 8px increments via GUI

bagelbyheart, 2021-03-17:
The Lakka GUI only allows adjustment of Fullscreen X and Y resolution by 8 pixel increments when using the GUI. In order to use resolutions that do not fit this pattern, you must either manually edit `.config/retroarch/retroarch.cfg` or set your resolution via the display driver (in the case of a Raspberry Pi, via `/flash/config.txt`.

Adjusting this would simplify user configuration, and would allow some handy performance hacks, such as running at 540p (see issue #1215) to maintain scaling and performance on more limited hardware.

> BEFORE you report a bug make sure you have tried the latest nightly version of Lakka at http://le.builds.lakka.tv/. Your bug might be already fixed.

The link to nightly builds appears to be out of date and should be https://nightly.builds.lakka.tv/latest/RPi2.arm/. Testing with the most [recent build](https://nightly.builds.lakka.tv/latest/RPi2.arm/Lakka-RPi2.arm-nightly-20210317-33af203.img.gz) the same behavior is seen. 

> If you are sure that it's a bug in Lakka, please answer the following questions:
> - Which version of Lakka are you using?

RPi2.arm-2.3.2 | Build Date Jan 20 2020

> - What system hardware are you using?

Raspberry Pi3

> - What did you do?

Attempt to set a fullscreen resolution of 960x540 using the GUI

> - What did you expect to happen?

I would select the appropriate x and y resolution from Settings > Video > Fullscreen Mode using either a number pad interface or a list of single pixel increments.

> - What happened instead?

I had to select a resolution that was a multiple of 8x, or apply the desired resolution by manually editing `.config/retroarch/retroarch.cfg`

```ini
video_fullscreen_x = "960"
video_fullscreen_y = "540"
```

ToKe79, 2021-03-18:
This feature is actually available - you have to enable advanced settings in Settings -> User Interface -> Show Advanced Settings. Then head back to Settings -> Video -> Fullscreen Mode -> here you will have to additional options - Fullscreen Width and Fullscreen Height.

Please, let me know if these options are mapped to the the same options in the config file.

Also, this is not a Lakka feature, but feature of RetroArch.

bagelbyheart, 2021-03-18:
Hey there @ToKe79 I just realized I opened this on the wrong project myself! Sorry about that. Taking the time to better formulate the feedback and see if I can't find the relevant code before I reopen there xD

As for the options you mentioned, those are what I initially used, the problem is that they are adjustable in 8px increments instead of 1 by 1 or (given there are already a few hundred entries per list at those increments) a number input field.

ToKe79, 2021-03-18:
@bagelbyheart and I forgot about the 8px increment you mentioned in the first sentence of your issue report when I got to the end. And I even opened up RetroArch to see if those options are in GUI and I did not test them myself - just saw them (with zero values, as I ran it in windowed mode on my laptop...).

Yes indeed, these options should be changeable by direct input of digits and not selecting a predefined values (for example as the option to insert Username under Settings -> User, but maybe limit only to digits).

Please open an issue/feature request at https://github.com/libretro/RetroArch.

ToKe79, 2021-03-18:
I looked at the source code of RetroArch - you can change the step in `menu/menu_setting.c`:
https://github.com/libretro/RetroArch/blob/4806a349206d23822da5cce35fb7f125c157cfb8/menu/menu_setting.c#L10468-L10481
For example this patch will allow for single steps of 4 pixel:
```
diff --git a/menu/menu_setting.c b/menu/menu_setting.c
index ca21b04ee1..b1bb66ef15 100644
--- a/menu/menu_setting.c
+++ b/menu/menu_setting.c
@@ -10477,7 +10477,7 @@ static bool setting_append_list(
                      general_write_handler,
                      general_read_handler);
                (*list)[list_info->index - 1].action_ok     = &setting_action_ok_uint_special;
-               menu_settings_list_current_add_range(list, list_info, 0, 7680, 8, true, true);
+               menu_settings_list_current_add_range(list, list_info, 0, 7680, 4, true, true);
                SETTINGS_DATA_LIST_CURRENT_ADD_FLAGS(list, list_info, SD_FLAG_ADVANCED);
 
                CONFIG_UINT(
```

bagelbyheart, 2021-03-18:
Thank you! I didn't realize that github search wasn't substring so it would have taken me forever to find this! I'll get the issue opened on the correct project after work today and will close this with a reference to that (or you can close it now)

closed 2021-03-23 by ToKe79