libretro/Lakka-LibreELEC #1227: [Feature request] Adding pip module to Python

DevilBlackDeath, 2021-03-26:
Hello,

I'm writing this feature request to possibly add the pip module for Python. I think Lakka would greatly benefit from extended Python support. The best way to do that without adding size overhead to Lakka and minimizing average user error would be to include the pip module that allows you to install nearly any Python module, giving you access to what you need if you need it, but using barely any storage if adding no other modules.

This would most importantly help with GPIO and overlay features, but other use cases can easily be thought of. Upgrading to Python 3 could also be a good idea, but is off-topic, and as a non-expert about Python I'm not sure how much complexity it would add to Lakka.

Thanks in advance for considering it !

DevilBlackDeath, 2021-03-27:
Just to add to this, I'm currently compiling Lakka while having the existing Python 3 package marked as additional package in Lakka's options file. Theoretically, with it being 3.7, pip should already be in. My compile is slow because it's done on a virtual machine, but I will report back on this once the compilation job is done !

60999, 2021-03-28:
Good!

DevilBlackDeath, 2021-03-28:
I was able to get what I intended to do by adding "python-raspberry-gpio" to the ADDITIONAL_PACKAGES variable in 'distributions/Lakka/options'.  As a quick "dirty" fix for at least Retroflag (and other similar cases) users I think adding it would be essential as to not lose the "average" non-developer Pi user. Any case with a poweroff switch handled through GPIO would benefit from that.

In the long run I still think adding pip is the best answer for most DIY projects involving Lakka (be it on the Pi or elsewhere, truth be told). That said, I don't know if upgrading Python to version 3 is viable. Apparently OpenELEC and LibreELEC have postponed it for years because a lot of their ecosystem relied on Python 2 (making the transition pretty hard). If any code in Lakka relies on Python 2, then upgrading to 3 could break stuff. Adding pip to Python 2 would be largely good enough though. A lot of Python developers seem to still try to maintain compatibility with Python 2, especially when it comes to packages targeting platforms that are more "exotic" in nature (like the Pi, Odroid...).

PR #1264 (2021-04-18): Added the RPi.GPIO Python library to the RPi Project
Following my feature request #1227 , after some tests is proves nearly impossible to have pip in Lakka. The efforts required to make it work makes adding python packages on a per-user basis easier. However RPi.GPIO is still a vital  package as it allows a lot of stuff, including safe shutdown scripts for cases with power buttons. While there are a lot of precompiled RPi.GPIO floating around, these only account for arm compilations, not aarch64. For such a core Pi Python component, I feel like having it compile with Lakka would be a benefit.
 [... cut]
closed 2021-04-18 by Ntemis