libretro/Lakka-LibreELEC #1232: Cores performance on RPi4

DevilBlackDeath, 2021-04-01:
`- Which version of Lakka are you using?`
3 or 4 different nightlies ranging from 03/24 to 04/01, also tested on 01/01 nightly. I tested no other nightlies between 2.3.2 and those ones, so I cannot confirm when the issue appeared.

`- What system hardware are you using?`
Raspberry Pi 4 Model B 2GB

`- What did you do?`
I tested Mupen 64 Plus Next on games that had decent or even good performances in 2.3.2 release.

`- What did you expect to happen?`
I expected better or similar performances.

`- What happened instead?`
I had very strong freezes and stutters.

I tested 3 games so far, Banjo-Kazooie, Banjo-Tooie and F-Zero. I tested as a SD boot and USB boot as well. I also tested on 2 different RPi 4. Finally I also tested both with the arm and aarch64 images.

Could be related to the sound muting issue #1177 reported in February, though I suffered no sound muting, only sound stuttering.

The one I need performances on is unfortunately part of the newer models on which Lakka 2.3.2 cannot run.

asdf288, 2021-04-02:
Mupen64Plus-Next versions from mid-2019 generally have very poor performance on the systems that I tested, I think this is not specific to the RPi 4. There are two particular commits from 2019 that crippled performance, making the core unusable on most lower spec devices.

If you're building Lakka yourself, you could try rebuilding the core using this commit hash from August 2019: b785150465048fa88f812e23462f318e66af0be0

DevilBlackDeath, 2021-04-02:
Huh ! Thanks for the hash ! I'll give that a try later. I thought I used 2.3.2 hash for a rebuild but now I think about I probably didn't. Are Mupen developers aware of that ? They were by far the best hope for 64 emulation on Pi4 now it seems aarch64 Parallel is probably a better source of hope. Tested it on nightlies, and while it does a regular hiccup (almost on a rhythm), outside of those hiccups it runs pretty well. Even runs the Banjo Kazooie intro that Mupen in 2.3.2 had issues with.

tl;dr : Thanks a lot for the suggestion, should I report that to Mupen NX devs or is it known (one of the devs seemed to imply it was not).

DevilBlackDeath, 2021-05-06:
For further clarification, this can now be closed. It's actually the defaults that make the core slow. Proper configuration actually makes it much faster than the old Mupen64 and Mupen64Plus AND more compatible. If it is at all possible to change a core's defaults for specific target platforms with the LibreElec's build system, then I would suggest changing a couple of things, namely :
- Switching Color buffer to Triple Sync
- Switching resolution to 320x240 (VERY FEW games can actually run at 640x480 on the Pi4)
- If base compatibility is less important than general performance, then I'd also recommend disabling depth buffer syncing, since very few games use it in meaningful ways (though performance gain is minimal to slight)

I don't know if that would require a custom mupen patch or anything like that, but these changes give better performances. The biggest one is 320x240 as 640x480 really takes a hit on the Pi4 (meaning a future Pi5 will likely have no problem at all handling 640x480, maybe even higher if they decide to add a dedicated GPU, hell a dedicated GPU, even super minimal, could probably push the 64 to 720p/1080p).

x2k13, 2021-06-23:
@DevilBlackDeath 
Do you have a smooth/working config to share for Mupen64Plus-Next for the RPi 3B+?
In 2.3.2 N64 games ran quite okay on the 3B+, but now not one single game runs on 3.2, see here as well:
https://github.com/libretro/Lakka-LibreELEC/issues/1332

dmrlawson, 2021-06-23:
> @DevilBlackDeath
> Do you have a smooth/working config to share for Mupen64Plus-Next for the RPi 3B+?
> In 2.3.2 N64 games ran quite okay on the 3B+, but now not one single game runs on 3.2, see here as well:
> #1332

Which version are you using on your 3b+, the rpi2 arm version or the new rpi3 aarch64 version?

x2k13, 2021-06-23:
> Which version are you using on your 3b+, the rpi2 arm version or the new rpi3 aarch64 version?

The latter. Fresh-Flash of 3.2 aarch 64, after trying the rpi2 32bit build, which has the same behaviour.

dmrlawson, 2021-06-23:
> The latter. Fresh-Flash of 3.2 aarch64, after trying the rpi2 32bit build, which has the same behaviour.

Ok, yes. Mupen is broken on this build, I'm trying to get it working. For now the rpi2 build is your best bet.

x2k13, 2021-06-24:
> > > > @DevilBlackDeath
> > > > Do you have a smooth/working config to share for Mupen64Plus-Next for the RPi 3B+?
> > > > In 2.3.2 N64 games ran quite okay on the 3B+, but now not one single game runs on 3.2, see here as well:
> > > > #1332
> > > 
> > > 
> > > Which version are you using on your 3b+, the rpi2 arm version or the new rpi3 aarch64 version?
> > 
> > 
> > The latter. Fresh-Flash of 3.2 aarch64, after trying the rpi2 32bit build, which has the same behaviour.
> 
> Ok, yes. Mupen is broken on this build, I'm trying to get it working. For now the rpi2 build is your best bet.
> 
> If that's not working either hopefully the cause is the same.

History of testing:
1. rpi2 arm 2.3.2 > works as expected, last working release
2. rpi2 arm 3.0 > crashes
3. rpi2 arm 3.1 > crashes
4. rpi2 arm 3.2 > crashes
5. rpi3 aarch64 3.2 > crashes

DevilBlackDeath, 2021-06-24:
@x2k13 Unfortunately I won't be of much help as I'm running it on the pi4. Crashes aside, The modifications I did were the main ones necessary for better performances unfortunately. If switching is not a problem then I'd recommend it, my tests were run on the cheapest pi4, the 2GB version, and I can tell you the games run extremely well for the most part. There's a bit of working around for some games (like Paper Mario or Rogue Squadron) and some still have areas of slowdown (Conker) and unfortunately there are still a few that aren't running at playable FPS (GoldenEye) but I did have some mostly 100% original hardware FPS surprises like Glover and Banjo Tooie (well Banjo Tooie slows down a bit during the Jiggywiggy challenges but that's pretty much it afaik).

dmrlawson, 2021-07-06:
@DevilBlackDeath closing this as I think you've solved the problem on rpi4, please reopen if you think I'm wrong about that.

Performance changes like that are likely due to updates to the core and I know with mupen specifically some GLideN64 updates slowed it down on the default configuration.

closed 2021-07-06 by dmrlawson