libretro/Lakka-LibreELEC #1242: Manual update not working for nightly builds

maxengel, 2021-04-06:
If you are sure that it's a bug in Lakka, please answer the following questions:
- Which version of Lakka are you using:
  - Lakka-RPi4.aarch64-nightly
- What system hardware are you using:
  - RPi 4 8GB
- What did you do?
  1. Downloaded the latest nightly build from (here)[https://nightly.builds.lakka.tv/latest/]
  2. Copied the img.gz file to the RPi via sftp to /Storage/.update
  3. Rebooted Lakka via the UI
- What did you expect to happen?
  - I expected the update to be applied successfully and for Lakka to reboot Retroarch into a functional UI
- What happened instead?
  - The console output indicated that the update was applied successfully
  - The RPi rebooted
  - The RPi only had a black screen and was inaccessible via SSH

davidgfnet, 2021-04-11:
I've been using this and seems to work. Perhaps running out of space? Or somehow a corrupted SD card or something?
I've used the update mechanism in several devices and seems to work well most of the time.

maxengel, 2021-04-11:
To verify, I just did the following:
1. On macOS 11.2.3 I downloaded the latest version of Etcher [1.5.116](https://github.com/balena-io/etcher/blob/master/CHANGELOG.md)
2. I wrote `Lakka-RPi4.aarch64-nightly-20210410-69d6e67.img.gz` to a different MicroSD card using Etcher that had previous been running Raspian fine
3. I installed the OS and then enabled SSH
4. I connected to Lakka via SSH and copied `Lakka-RPi4.aarch64-nightly-20210411-bfecf09.img.gz` to the `.update` directory
5. I rebooted Lakka
6. The upgrade went fine and then the RPi rebooted at the end
7. The screen on my display was all black after reboot but had an active signal
8. Lakka failed to boot, and was inaccessible via SSH or SMB, nor could it be pinged by the static IP associated with it

I've been using the 64-bit nightlies for over a month without issue on the same RPi with the same SD card. However, this started recently happening and is reproducible with a different SD card. I'm happy to do further debugging and/or provide files if it helps.

maxengel, 2021-04-11:
@davidgfnet: I found a fix, but don't know enough about LibreELEC/Lakka to submit a solution:
1. After an upgrade, the `distroconfig.txt` file is missing the following as the end of the file:
```
arm_64bit=1
kernel=kernel.img
```

Those lines are present after the initial flash of the Micro SD card, so something in the upgrade process must not be generating those two lines.

davidgfnet, 2021-04-11:
So adding those two lines back fixes the issue?
Thanks for debugging!

maxengel, 2021-04-11:
Happy to help!

Yes, sorry for not clarifying. If I add those two lines back that are missing after the upgrade everything boots normally.

jdalmanza, 2021-04-11:
i had the same experience on the 4/6 nightly, lakka did not boot after post upgrade reboot. using pi4 4gb and aarch64 image

PR #1258 (2021-04-12): Move bootloader config patch for consistency
This makes all copies of the file identical. This hopefully fixes #1242
commit 70a35525 (2021-04-12): Move bootloader config patch for consistency (#1258)

This makes all copies of the file identical. This hopefully fixes #1242
closed 2021-04-12 by Ntemis