libretro/Lakka-LibreELEC #1244: xbox one control model 1914 does not connect via bluetooth

z-our, 2021-04-07:
I currently have 2 xbox one controls, one is model 1708 which connects perfectly in lakka via bluetooth.

But the new xbox one 1914 control model is not possible to connect via bluetooth

I do the same steps I do to connect the 1708 but the 1914 control keeps blinking.

gouchi, 2021-04-21:
Hi,

Can you try if it is working with another Linux distribution ?

You may need this [driver](https://github.com/medusalix/xow#supported-devices).

Thank you for your tests.

Niels778, 2021-06-04:
I have the same controller model, and the same issue.
However, not possible to update driver in lakka (also nightly build).

Or am I mistaken?
thx!

Ntemis, 2021-06-04:
Early xbox one controllers need a special usb dongle to connect

Niels778, 2021-06-05:
No, this is the latest model. And also with dongle, it is not working.

Niels778, 2021-06-05:
Please open the issue again?

Ntemis, 2021-06-06:
Xow cant be used due to license issues as Microsoft's Terms of Use, which strictly disallow the distribution of the firmware.
Closing down

closed 2021-06-06 by Ntemis