libretro/Lakka-LibreELEC #1256: ParaLLEl-n64 - CXD4 RSP plugin crashes RetroArch on AArch64 RPi4

autofire372, 2021-04-11:
This issue tracker is NOT for help or support. Please use one of the following channels for help and support:
- the Lakka forum (https://forums.libretro.com/c/libretro/lakka-tv-general)
- via IRC @ #lakkatv on freenode.net
- #lakkatv on libretro discord https://discord.gg/retroarch

Before you report a bug make sure you have tried the latest nightly version of Lakka at https://nightly.builds.lakka.tv/latest/. Your bug might be already fixed.

If you are sure that it's a bug in Lakka, please answer the following questions:
- Which version of Lakka are you using?
- What system hardware are you using?
- What did you do?
- What did you expect to happen?
- What happened instead?

It is important to provide logs for debugging as part of your report. Instructions for generating logs can be found in the Troubleshooting Lakka documentation at https://www.lakka.tv/doc/Troubleshooting-Lakka/

Please delete this text before posting.

Lakka version: latest AAarch64 nightly
Hardware: Raspberry Pi 4

Attempting to run any game with the parallel-n64 core with the cxd4 rsp plugin active causes RetroArch to crash with the following error:

`[libretro ERROR] mupen64plus: Compile at bogus memory address: a4001550`

This error was captured by adding `retroarch=0` to cmdline.txt, then SSHing into Lakka and running `retroarch -v`.

Ntemis, 2021-04-11:
@dmrlawson is Rpi4 up to specs for this core?

dmrlawson, 2021-04-11:
I'd say not. It barely performs with HLE and these are all LLE plugins.

It'd be nice if it didn't crash though. One thing to note is that parallel-rdp (the vulkan plugin) requires Vulkan 1.1 whereas we only have 1.0 on rpi4 so far, so there's no chance of this working atm.

autofire372, 2021-04-11:
I admit I was trying parellel-n64 out of desperation, as mupen-next has suffered severe performance regressions (as noted in #1232) that render even Super Mario 64 unplayable, when that core used to allow rpi4 to handle anything that wasn't Goldeneye or Perfect Dark.

closed 2021-04-11 by Ntemis