libretro/Lakka-LibreELEC #1260: lakka-update fails (Wrong URL?)

autofire372, 2021-04-13:
From ssh:

```
Lakka (official): devel-20210410013440-12abb6f (RPi4.aarch64)
Lakka:~ # lakka-update
wget: server returned error: HTTP/1.1 404 Not Found
:: Downloading upgrade
Connecting to le.builds.lakka.tv (195.201.56.250:80)
Connecting to le-builds.lakka.tv (172.67.197.143:80)
wget: server returned error: HTTP/1.1 404 Not Found
```

Shouldn't it be looking for "nightly.builds.lakka.tv"?

davidgfnet, 2021-04-13:
I recall this was disabled... Perhaps we could re-enable though.
What do you get running: cat /etc/release | cut -d - -f 1 
?

ToKe79, 2021-04-13:
it is enabled only for nightly builds, not from PR builds.

ToKe79, 2021-04-13:
if you are on PR build, just `cd ~/.update && wget <url> && reboot`

closed 2021-04-13 by ToKe79

davidgfnet, 2021-04-13:
How do you know it's a PR build? Is it due to the "20210410013440" ?