libretro/Lakka-LibreELEC #1261: [Feature Request] Option to eject the optical drive

tiagolascasas, 2021-04-14:
Hi, I have a bit of an odd request...

I'm using Lakka on an x86_64-based PC with an optical drive. It works fine with the latest version of Lakka, but unfortunately my optical drive is really old and the eject button does not work, which requires me to eject it through software (on Windows, for example, this would be done by right-clicking the drive and clicking "Eject"). 

On Lakka, though, the only way of doing this is to ssh into the machine and type in the `eject` command, which is really inconvenient. It'd be nice if there was an extra option to eject it through Lakka itself, maybe alongside the "Load Disc" and "Dump Disc" options. I know this would probably be a super low priority QoL improvement for you guys, but still, it'd solve my problem.

Thanks in advance!

ToKe79, 2021-04-16:
Hi,
can you test image from [here](https://nightly.builds.lakka.tv/members/vudiq/Generic.x86_64-EJECT/) if [this](https://github.com/ToKe79/RetroArch/commit/314727456586122a13ef9bdc3e4baeacd48f5b21) implementation works?
Thank you!

gouchi, 2021-04-18:
I confirm it is working ;)

PR #1270 (2021-04-18): retroarch: eject + timezone patch
fixes #1261 and https://github.com/libretro/RetroArch/issues/5183

# CD ROM eject
adds new entry (where the entries for dumping CD is) to eject the disc - in case the CD drive has no physical button to eject the disc / the button does not work

# Time zone
adds new entry to Services sub-menu of Settings (where other Lakka specific settings are - enable/disable Samba/SSH/...) and allows the user to set the time zone from the GUI without needing access to the command line / file system
closed 2021-04-18 by Ntemis