libretro/Lakka-LibreELEC #1263: Broken compilation on x86_64

Yqnn, 2021-04-15:
Hello,

Since March 23, I'm not able to compile Lakka anymore for x86_64.
It fails on `icu:host` with the following error:
```
/usr/bin/ld: /media/data/build/Lakka-LibreELEC/build.Lakka-Generic.x86_64-3.0-devel/toolchain/lib/libicuuc.a(putil.ao): relocation R_X86_64_PC32 against symbol `__timezone@@GLIBC_2.2.5' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: final link failed: bad value
collect2: error: ld returned 1 exit status
make[2]: *** [Makefile:161: ../lib/libicui18n.so.68.1] Error 1
make[2]: Leaving directory '/media/data/build/Lakka-LibreELEC/build.Lakka-Generic.x86_64-3.0-devel/icu-68.1/.x86_64-linux-gnu/i18n'
make[1]: *** [Makefile:153: all-recursive] Error 2
make[1]: Leaving directory '/media/data/build/Lakka-LibreELEC/build.Lakka-Generic.x86_64-3.0-devel/icu-68.1/.x86_64-linux-gnu'
FAILURE: scripts/build icu:host has failed!
[259/353] [FAIL] build   icu:host

The following logs for this failure are available:
  stdout: /media/data/build/Lakka-LibreELEC/build.Lakka-Generic.x86_64-3.0-devel/.threads/logs/262/stdout
  stderr: /media/data/build/Lakka-LibreELEC/build.Lakka-Generic.x86_64-3.0-devel/.threads/logs/262/stderr

parallel: This job failed:
package_worker 2 262 353 'build icu:host'
Parallel build failure - see log for details. Time of failure: Thu Apr 15 08:56:13 CEST 2021
make: *** [Makefile:12: image] Error 1
```

I'm using the following command:
`DISTRO=Lakka PROJECT=Generic ARCH=x86_64 BUILD_PERIODIC=nightly  make image`

Thanks!

Yqnn, 2021-04-15:
Seems that putting back the `--enable-static` and `--disable-shared` options fix the issue.
(they were removed in bfe3efb8a2887c427e182c02071c398df2dce148  )

Ntemis, 2021-04-15:
@asdf288 ^^

ToKe79, 2021-04-15:
This is not a bug, you have to remove old libs and binaries from toolchain and sysroot as mentioned in the related pull request: https://github.com/libretro/Lakka-LibreELEC/pull/1221#issuecomment-803656991
Then also revert your local changes to icu package. For Lakka we need shared library, LibreELEC uses icu only for addons where they use static libraries. Also clean build folders for liblcf, icu and easyrpg (e.g. `DISTRO=Lakka PROJECT=Generic ARCH=x86_64 ./scripts/clean easyrpg`).

closed 2021-04-15 by ToKe79