libretro/Lakka-LibreELEC #1266: PPSSPP asset folder and crashing on RPI 4. Nightly Builds

Rstation4, 2021-04-16:
Lakka OS:   Lakka-RPi4.arm-nightly-20210415-12a3881
Raspberry Pi 4 4gb

I first did a run without adding the PPSSPP asset folder from Hrydgard github https://github.com/hrydgard/ppsspp and ran an iso rom. It worked and there was no issues in that short run. So I tried it again but this time with a ppsspp asset folder I had and it ended up acting funny with the fonts getting kinda scrambled and corrupted. On my third try I deleted the folder and used a fresh download form Hrydgard and it simply did not even launched, it went to a blank black screen and back to the xmb menu.

For the 4th try I did a clean install of Lakka and added the hrydgard PPSSPP folder, A DLC content into /system/PPSSPP/GAME. I ran it and it did the same result with crashing with a black blank screen.

Has there been any changes with the core that it is no longer needed?

Logs: 
[Retroarch Lakka ppsspp.log](https://github.com/libretro/Lakka-LibreELEC/files/6321693/Retroarch.Lakka.ppsspp.log)

[image]

ToKe79, 2021-04-16:
Hello,
it is not necessary to manually add the ppsspp-assets folder - we already ship the data in the `SYSTEM` partition and overlay them in `/tmp/system`. Please, retest without adding the ppsspp-assets manually.
However, as we now use overlayfs for the `system` folder, it might be necessary to reboot after you add your files to /storage/system to get them applied to the `/tmp/system` overlayed folder (not sure). Another solution is to directly copy your files to `/tmp/system`, in such case do not transfer your data to `/storage/system` (or `System` share on samba), but to a different folder and then copy the files over using command line access (e.g. ssh). Or use sftp client to copy directly to `/tmp/system`.

Rstation4, 2021-04-16:
Where would DLC content for the PSP go? I remember the old instructions would be in PPSSPP/GAME/

ToKe79, 2021-04-16:
if you look at `/tmp/system`, there should be already `PPSSPP` folder with the assets, so try copying there / create `GAME` folder under `/tmp/system/PPSSPP`. physically it will be stored to /storage/system - the overlayfs should take care of that.

Rstation4, 2021-04-16:
Thank you I resolved it by deleting that asset folder I placed in /system. Even as root I cant get access to /temp nor /assets but I did not need to in the end. I think the documents over at lakka need to be updated. For my DLC issue, I resolved it by placing it in the path. 
LAKKA_DISK/savefiles/PSP/SAVEDATA   (This is where I needed to place them for "Mega Man Powered Up")

but for another game specifically Metal Slug XX it need to be in
LAKKA_DISK/savefiles/PSP/GAME

I hope this info helps out anyone trying to add DLC content for their PSP.

closed 2021-04-16 by ToKe79