libretro/Lakka-LibreELEC #1268: [Suggestion] Keyboard auto-detect (RPi 4, nightly build Lakka)

Rstation4, 2021-04-17:
Raspberry Pi 4 B 4gb
Lakka Nightly  RPI 4 arm Apr 15 2021  

Hardware: USB Keyboard and USB Mouse
Cores: DosBox and DosBox-SVN

Im not sure if it is a bug, but when I started a first time setup and attempt at DOSBox the keyboard was not allowing me to input certain keys. IE P was pausing, I quickly had to delete all the keyboard hotkeys, then when Enter and some other keys such as the arrows and x was not inputting I went to controls to change them from gamepad to keyboard and mouse for the core. 

It did not allow me to use those keys until  I disconnected the gamepad so RA  will not get confused, Using the Keyboard I had to  change it from gamepad to keyboard+ mouse and then delete the keys form the input list.

ToKe79, 2021-04-17:
Can you, please, check, if same behavior is in standalone RetroArch in e.g. Raspberry Pi OS? If so, then this has to be reported upstream, not here.

Rstation4, 2021-04-17:
So I did end up testing it out on RetroArch which is also installed on my Linux Mint 20.1 Desktop PC. It does the same thing, I never used Dosbox on it but I just downloaded the core and ran a game with it, with my gamepad, Mouse and Keyboard hooked up. 
RA version of PC: 1.9.1
Build Date: March 30, 2021

closed 2021-04-17 by Rstation4

ToKe79, 2021-04-17:
More about DOSBox input in RetroArch - https://docs.libretro.com/library/dosbox/#user-1-2-device-types
Maybe changing the Device Type (?) or other options will get the desired efect?

Rstation4, 2021-04-18:
One of the suggestions I have gotten from another member on the RetroArch Github page.

"_Starting in v1.9.1, there's settings > input > Auto Enable 'Game Focus' Mode (may be an advanced setting, in which case you need to go to settings > user interface > show advanced settings ON before it will be visible), which will automatically trigger the 'game focus mode' setting (mapped to "scroll lock" key by default). This is essentially keyboard/mouse capture, insofar as it sends those events directly to the core instead of listening for hotkeys, etc._"