libretro/Lakka-LibreELEC #1275: Segfaults in libLLVM-3.9.so, r300_dri.so and retroarch

gitendo, 2021-04-23:
- Which version of Lakka are you using?
Lakka-Generic.x86_64-nightly-20210423-d52f851.img.gz

- What system hardware are you using?
x86_x64 [HP T5730](https://www.parkytowers.me.uk/thin/hp/t5730/)

- What did you do?
Tried to boot live version (nightly build). Boot / install 2.3.1 and 2.3.2.

- What did you expect to happen?
To see how it looks / works.

- What happened instead?
Flickering screen with Lakka logo on nightly and black screen on stable builds. Doesn't work.

Logs x86_64-nightly-20210423-d52f851:
[dmesg.txt](https://pastebin.com/HuCtN2r0)
[log.txt](https://pastebin.com/EnqESn9N)
[snd.txt](https://pastebin.com/2dtGRbJml)
[vga.txt](https://pastebin.com/Dtvujgyn)

Logs Generic.i386-2.3.1:
[dmesg.txt](https://pastebin.com/R4ckCmSE)
[log.txt](https://pastebin.com/17uBzXpZ)

gouchi, 2021-04-23:
Hi,

Thank you for the logs.

You should try first with Lakka 2.3.1 version because it seems your card is too old for xmb menu.

Please try to edit `menu_shader_pipeline = "0"` in [retroarch.cfg](http://www.lakka.tv/doc/Configuring-Lakka/). If it is still not working, try to use rgui menu driver with `menu_driver = "rgui"`

davidgfnet, 2021-04-23:
Could you run lsmod and glxinfo and paste the logs?
The nightly should work IMHO. The stable Lakka might not, its using mesa 17 and ive found several reports in Mesa 17&18 that have bugs for that driversl. But mesa 21 should support that gpu right?

gitendo, 2021-04-24:
@gouchi 
I've added these options but the only result was black screen flickering stopped and I couldn't connect via ssh anymore.
```
Lakka (official): 2.3.1 (Generic.i386)
Lakka:~ # nano /storage/.config/retroarch/retroarch.cfg
```

@davidgfnet 
Night build at least shows flower logo and build number in top left corner. But then screen goes black for 2-3 seconds, returns to logo and again.  Here's [lsmod log](https://pastebin.com/vncFtx6X). As for glxinfo - it's missing and I guess Lakka doesn't have package manager to download/install it?

davidgfnet, 2021-04-24:
lsmod doesnt show the graphics driver for AMD.bthat explains why Reotrarch cant load  a GL drver and crashes. Can you run perhaps a lspci?
And im not sure whether we ship the driver or not, but you could try to "modprobe radeon" and see what happens (dmesg might print some info if modprobe fails)
I see that this card has some compat issues in other distros. Does any other distro work well with it? If so, which ones?
Thanks!

Ntemis, 2021-04-24:
Your issue is plain and simple compatibility. Your card is so old it doesn't support OpenGL v3.x so a special build is needed for it to work

gitendo, 2021-04-25:
@davidgfnet 
Sure. Here's [lspci log](https://pastebin.com/6M4TbnEQ). `modprobe radeon` doesn't make any difference since it looks like there's no radeon module to be found - [modules list](https://pastebin.com/AAePSrGe). As far as I remember HP had some Debian version dedicated for it but I've quickly booted [Zorin-OS-15.3-Lite-64-bit](https://zorinos.com/download/#lite) and [no problems](https://i.imgur.com/sZoA60G.jpg) here. It has radeon driver working. I was also able to boot [Recalbox](https://www.recalbox.com/).

gouchi, 2021-04-25:
I missed this log for the 2.3.1

> Unable to find target for this triple (no targets are registered)

There is a workaround launching RA with 

`systemctl stop retroarch.service && DRAW_USE_LLVM=0 retroarch`

[Source](https://github.com/libretro/Lakka-LibreELEC/issues/618#issuecomment-472534970)

davidgfnet, 2021-04-25:
Does recalbox have the radeon module?
I have the feeling the lack of suck module is the root cause, since no amount of GL drivers and mesa will be useful without the kernel module that exposes the dri device.

gitendo, 2021-04-25:
@gouchi 
```
Lakka:~ # systemctl stop retroarch.service && DRAW_USE_LLVM=0 retroarch
Segmentation fault (core dumped)
```
And more verbose output:
```
Lakka:~ # systemctl stop retroarch.service && DRAW_USE_LLVM=0 LIBGL_DEBUG=verbose retroarch --menu --verbose
[INFO] RetroArch 1.7.8 (Git ea7e682253)
[INFO] === Build =======================================
[INFO] CPU Model Name: Mobile AMD Sempron(tm) Processor 2100+
[INFO] Capabilities: MMX MMXEXT SSE1 SSE2 SSE3
[INFO] Built: Sep 16 2019
[INFO] Version: 1.7.8
[INFO] Git: ea7e682253
[INFO] =================================================
[INFO] Environ SET_PIXEL_FORMAT: RGB565.
[INFO] Redirecting save file to "/storage/savefiles/.srm".
[INFO] Redirecting savestate to "/storage/savestates/.state".
[INFO] Version of libretro API: 1
[INFO] Compiled against API: 1
[INFO] [Audio]: Set audio input rate to: 30000.00 Hz.
[INFO] [Video]: Video @ fullscreen
[INFO] [Video]: Starting threaded video driver ...
[INFO] [DRM]: Found 2 connectors.
[INFO] [DRM]: Connector 0 connected: no
[INFO] [DRM]: Connector 0 has 0 modes.
[INFO] [DRM]: Connector 1 connected: no
[INFO] [DRM]: Connector 1 has 0 modes.
[WARN] [DRM]: Couldn't get device connector.
[WARN] [DRM]: Couldn't get device resources.
[ERROR] [KMS]: Couldn't find a suitable DRM device.
[INFO] [GL]: Found GL context: null
[INFO] [GL]: Detecting screen resolution 320x240.
function is no-op
function is no-op
function is no-op
function is no-op
[INFO] [GL]: Vendor: (null), Renderer: (null).
[INFO] [GL]: Version: (null).
Segmentation fault (core dumped)
```
I think David is right, there's no radeon module.

@davidgfnet 
Yes, it does: 
```
recalbox-x86.img.xz\recalbox-x86.img\boot\recalbox\lib\modules\5.8.18\kernel\drivers\gpu\drm\radeon\radeon.ko`
recalbox-x86.img.xz\recalbox-x86.img\boot\recalbox\lib\modules\5.8.18\kernel\drivers\video\fbdev\aty\radeonfb.ko`
```

davidgfnet, 2021-04-25:
I checked the config and the radeon module is builtin, so its normal that it doesnt show up in lsmod nor as a .ko file.
Check /sys/module, should be there!

gitendo, 2021-04-25:
You're right, I didn't know that.

davidgfnet, 2021-05-27:
I see there's a few bugs here and there on the same driver:
https://gitlab.freedesktop.org/mesa/mesa/-/issues/386
Perhaps we should try to get a proper debug coredump and report it upstream, otherwise it might never get fixed ^^'
What other Linux distros have you tried on that machine?

gitendo, 2021-05-31:
Thanks David, but I think I'll give up. I thought this might be easily solved, so I could check that thin client's performance wit Lakka.
Looks like it's easier to stick to Win XPe and use AttractMode instead.

closed 2021-05-31 by gitendo