libretro/Lakka-LibreELEC #1285: no sound/mouse on chromebook lenovo 500e

hugyry, 2021-05-07:
This issue tracker is NOT for help or support. Please use one of the following channels for help and support:
- the Lakka forum (https://forums.libretro.com/c/libretro/lakka-tv-general)
- via IRC @ #lakkatv on freenode.net
- #lakkatv on libretro discord https://discord.gg/retroarch

Before you report a bug make sure you have tried the latest nightly version of Lakka at https://nightly.builds.lakka.tv/latest/. Your bug might be already fixed.

If you are sure that it's a bug in Lakka, please answer the following questions:
- Which version of Lakka are you using?
- What system hardware are you using?
- What did you do?
- What did you expect to happen?
- What happened instead?

It is important to provide logs for debugging as part of your report. Instructions for generating logs can be found in the Troubleshooting Lakka documentation at https://www.lakka.tv/doc/Troubleshooting-Lakka/

Please delete this text before posting.

gouchi, 2021-05-21:
As described in the [issue template](https://github.com/libretro/Lakka-LibreELEC/blob/master/ISSUE_TEMPLATE.md), please try to provide some [logs](http://www.lakka.tv/doc/Troubleshooting-Lakka/) at least `audio` and `dmesg` log.

Thank you.

Ntemis, 2021-06-06:
no response closing down

closed 2021-06-06 by Ntemis