libretro/Lakka-LibreELEC #1286: performance regression.  default cpu governor is now power saver

auxfam, 2021-05-07:
This issue tracker is NOT for help or support. Please use one of the following channels for help and support:
- the Lakka forum (https://forums.libretro.com/c/libretro/lakka-tv-general)
- via IRC @ #lakkatv on freenode.net
- #lakkatv on libretro discord https://discord.gg/retroarch

Before you report a bug make sure you have tried the latest nightly version of Lakka at https://nightly.builds.lakka.tv/latest/. Your bug might be already fixed.

If you are sure that it's a bug in Lakka, please answer the following questions:
- Which version of Lakka are you using?
nightlies: 818694606b03ac5f3d0fc1114cf6f0e39e382a3b [ Commits on May 3, 2021 ]
- What system hardware are you using?
intel NUC8  Intel® Core™ i5-8259U Processor (6M Cache, up to 3.80 GHz) 
- What did you do?
installed latest nighties as above
run swanstation
with gl backend
+ crt-pi shader (one of the lighter resource ones)

- What did you expect to happen?
full 60 fs no drops when running tekken 3
+ scanlines/crt shader look as stated above

- What happened instead?
many folder more stuttering, and dropping down to around 40% frame loss (60 -> 40/30)
disabling crt shader solvers this.

- other notes
under the retroarch option:ppwermanagement -> cpu governer.  appears to be set to `power saver` by default and when flipping between this and the game I notice it barely moves from 600Mhz  so the system at the very least believes it is idling.    setting this to `performance` for all course, shoot up the indicated lockspeed to 3800Mhz and the stuttering goes away entirely

also tried this with some heavier cores (Citra/Doplhin)
`power saver`: boosts a little over 2000MHz  but drops really easily.   in most titles stutters and running around 85% speed
`performece`:  stays high around 3800Mhz,  testing the same titles run much faster (full sped in most cases) and random stutters are gone (except for shader compilation which is similarly present across both)

current workaround is to start up lakka -> go into power management -> flip all core governors to `performance`.   this is not saved so each startup you'd have to do this.

similar stuttering happens on 16/8bit cores if you enabled frame delay as well

davidgfnet, 2021-05-07:
Hey! Its odd cause this commit does not change the default governor, it just allows users to tweak it.
I checked the default linux configs and the governor should be performance for x86 and schedutil for x64.

auxfam, 2021-05-08:
this is the build I used:  https://nightly.builds.lakka.tv/2021-05-04/Generic.x86_64/Lakka-Generic.x86_64-nightly-20210504-8186946.img.gz

I do not doubt that the kernal options are set to performence on boot.  However post boot go to `power management->cpu governer` and you should see it has flipped to `power saver` and (if the readings shown there are accurate) each cpu core sitting at whatever the minimum is (600MHz in the NUC8i5's case)

switching to `performence` under the `power management->cpu governer` section and booting up Super Street Fighter IV for Citra/3DS at 2x resolution + max 3 frame buffer gives me full speed on actual matches (no drops) wheas leaving it on defaults on a fresh boot I get 55-57 and frequent drops and audio stutter

on Snes9x cores I manage to reduce max buffer chain to 2 + 10ms frame delay wheas previously I could barefly get 6 on many titles.

in any case there is a not tirvial segment of perfomence being left on the table here for users of mid-high (> 3GHz) end (x86_64) intel cpus for sure unless users go into `power management->cpu governer` and switch to `performence` on each fresh boot.  Either that or I imagine disabling C-States in Bios might force the same thing (but not all boards support such options)

auxfam, 2021-05-17:
an update:  

just installed the changes from `Improved CPU governor/frequency scaling control menu`

it works.  the default menu/core powersave/performance setting is a good improvement in frame times overall  and makes for a good default.

in terms of performance I retested this with SFIV and if fall ever so slightly behind plain `performance` and `all core performance` profiles if I were to guess it's around 95% or so.  minor drops/stutters when performing fireballs et all on occasion (none if you leave the profile on performance entirely so I guess some backend thread causing some minor cpu throttling)

in general the custom managed profile (menu/core -> ondemand/performance) seems to be a good balance as long as you configure the min cpu clockspeed to be around 60% whatever your CPU turbo is

all core performance > custom managed profile > managed profile

all in all I think this was a great addition for non mobile / PC users

thanks for the great work here

davidgfnet, 2021-05-17:
Performance should always pick the highest frequency (no matter what the minimum is). So it should not exhibit any difference.
Anyway, since this is configurable now (as opposed to hardwired to some per-platform default) we can close this and let users experiment and pick.
Thanks

closed 2021-05-17 by davidgfnet