libretro/Lakka-LibreELEC #1290: Raspberry Pi Composite (3.5mm) output broken

jayare5, 2021-05-11:
There was a PR that was expected to fix it here but it didn't:

https://github.com/libretro/Lakka-LibreELEC/pull/1273

@dmrlawson, I tested both Pi3 and Pi4 (with enable_tvout=1) and they don't work.
However, I know they're outputting something because the TV detects the signal, but it's black.
Using sdtv_mode=16 actually works, and gives a 240p signal, so I know the Pi4 has enable_tvout, but it's all black.

Thanks!!

dmrlawson, 2021-05-11:
I think I've got a cable I can use to look into this, I just need to find some time.

I suspect it's one of:
 - missing kernel config
 - missing/wrong files in `/boot`
 - something not correctly selecting the right output

Hiki1, 2021-05-22:
I've retested this issue on Pi3 and Pi4 with Lakka 3.0 stable and unfortunately it still does occur.

gouchi, 2021-05-23:
- Link to [upstream LE issue](https://github.com/LibreELEC/LibreELEC.tv/issues/5335).
- Source to the [fix](https://github.com/raspberrypi/linux/pull/4240/files).

dmrlawson, 2021-05-23:
> * Link to [upstream LE issue](https://github.com/LibreELEC/LibreELEC.tv/issues/5335).
> * Source to the [fix](https://github.com/raspberrypi/linux/pull/4240/files).

We're using a commit later than that in Lakka 3.0, do we still need to enable something?

gouchi, 2021-05-23:
It seems it needs some testing according to this [thread](https://forum.libreelec.tv/thread/24014-composite-video-in-le10-on-rpi4/).

I don't know if something else is needed more than RPi linux kernel and firmware updated and `enable_tvout=1`.

Ntemis, 2021-06-06:
@jayare5 please test 3.1 release

jayare5, 2021-06-07:
@Ntemis Tested! Unfortunately still broken.

gouchi, 2021-06-08:
I see that there is also this [PR](https://github.com/raspberrypi/linux/pull/4241) , not sure if it is needed for this issue.

jayare5, 2021-06-08:
@gouchi Can it be used to test?

gouchi, 2021-06-08:
You can make a test with this [image](https://nightly.builds.lakka.tv/members/gouchi/1290/Lakka-RPi4.arm-3.0-devel-20210608232256-449e51e.img.gz) for RPi4.

Thank you.

jayare5, 2021-06-09:
@gouchi Tested, it still displays only black.

Notes: Since this is Pi4, I used enable_tvout=1, with this I can tell that it's outputting black because the TV flickers for a bit when it detects the signal, so I know the option is working, just no image or video coming.

Another interesting note: Normally when you power on the RPi, you can only use either Composite or HDMI, never both at the same time, and it requires you to power off if you want to switch. With this test image, even though I powered on the RPi connected to my CRT with the AV cable, I connected the HDMI cable to check and it was actually outputting through HDMI at 1080p! RetroArch is not even supposed to boot at 1080p when connected to a CRT with the 3.5mm AV cable. Not sure if that says anything, just something I noticed.

gouchi, 2021-06-09:
@jayare5 Thank you for your tests. Can you try  ?

> And I can confirm that then, enable_tvout=1 and dtoverlay=vc4-kms-v3d-pi4,composite=1 in config.txt result in usable picture in X on Pi4.

It might help for our case.

[Source](https://github.com/raspberrypi/linux/issues/4233#issue-838158441)

jayare5, 2021-06-09:
Adding that line makes the Lakka berry show up on the TV!
But it doesn't seem to get past it, at least for the 3 minutes I left it on.

gouchi, 2021-06-09:
Thank you for your tests.

Will it possible to get some [RA logs](https://www.lakka.tv/doc/Troubleshooting-Lakka/#lakka-system-logs) if possible ?

jayare5, 2021-06-10:
@gouchi I followed the guide and have a log.txt file!
Not sure if this is correct though since this is my first time creating a log, let me know.
[log.txt](https://github.com/libretro/Lakka-LibreELEC/files/6627838/log.txt)

gouchi, 2021-06-12:
Thank you for the log.

It seems you got segfault :(

Can you try with [nightly builds](https://nightly.builds.lakka.tv/latest/) to boot with `enable_tvout=1` and `dtoverlay=vc4-kms-v3d-pi4,composite=1` in `config.txt`  and provide some log ?

jayare5, 2021-06-14:
Done! Unfortunately, I got basically the same issue, seeing as the log looks practically the same except for newer nightly version.
The Lakka berry did not show up (as expected) but I did use those 2 options in the config.txt.

To make sure everything is correct, I connected to HDMI and saw that the SSH window looked good with all the info showing and no segment fault, but I just get segment fault again when it's connected with the composite cable.

[log (2).txt](https://github.com/libretro/Lakka-LibreELEC/files/6646869/log.2.txt)

gouchi, 2021-06-16:
Thank you for your tests.

We will have to wait a little maybe some update from RPi linux kernel.

jayare5, 2021-06-16:
No problem! Here's hoping :)

gouchi, 2021-07-21:
@jayare5 can you try also to a make test adding those options in `config.txt `?

```
hdmi_force_hotplug=0
hdmi_ignore_hotplug=1
enable_tvout=1
```

> hdmi_force_hotplug
> Setting hdmi_force_hotplug to 1 pretends that the HDMI hotplug signal is asserted, so it appears that a HDMI display is attached. In other words, HDMI output mode will be used, even if no HDMI monitor is detected

> hdmi_ignore_hotplug
> Setting hdmi_ignore_hotplug to 1 pretends that the HDMI hotplug signal is not asserted, so it appears that a HDMI display is not attached. In other words, composite output mode will be used, even if an HDMI monitor is detected.

[Source](https://www.raspberrypi.org/documentation/configuration/config-txt/video.md)

jayare5, 2021-07-23:
Hmmm so I'm not sure if I'm doing something wrong, but there are no differences using this!
Nothing happens if I just add this to config.txt.
I tried latest nightly as well as an old one that was installed before flashing a new nightly.

The Composite output will only give me black on my CRT, but plugging in he HDMI cable will display on the HDMI screen. (not supposed to happen)
If I use dtoverlay=vc4-fkms-v3d-pi4, I just see the Lakka berry but stays there frozen. With this one I do not see HDMI working after it's already turned on.

gouchi, 2021-07-23:
Thank you for your tests I wished it could have improved the situation.

It should have at least disable the hdmi output.

jayare5, 2021-07-23:
Hmmm it definitely didn't do that! What are the chances I did something wrong?
I had actually also checked that the same config wasn't anywhere else on the config.txt, well except for one of the hotplug ones which is there commented by default, which should have done nothing.

I'll double check though. Let me know what else I should look out for.

eugene-s-nesdev, 2021-07-24:
I confirm this problem on RPi3B+ with lastest "Lakka-RPi3.aarch64-nightly-20210724-58dafbe.img" 
- i've installed fresh lastest build and tested it on 1920x1080 display via HDMI - all works good.
- add this lines to config and plug RPI to CRT-TV via 3.5mm out:
```
hdmi_force_hotplug=0
hdmi_ignore_hotplug=1
sdtv_mode=16
```
Signal is present (i see deep dark-gray backgroung with some noise) but there are no picture.

Lakka 2.2.x/2.3.x works excellent on CRT with same settings

Ntemis, 2021-07-24:
Report this to raspberry pi kernel repo so we can fix this in the future

eugene-s-nesdev, 2021-07-25:
CRT-test update:
- Lakka 2.2.x works 
- 2.3.x hangs at flower
- 3.x show blackscreen.
2.2.x is removed from the server archive: http://le-builds.lakka.tv/

eugene-s-nesdev, 2021-08-06:
That lakka-nightly which i've tested had 5.10.50 kernel https://github.com/libretro/Lakka-LibreELEC/commit/eb4a4fe0a29de5afae4e92ee2ea7bb6cd9c22b8e
but still don't working. So I need to test RPiOS:

I've downloaded "2021-05-07-raspios-buster-armhf.img", tested it on LCD via HDMI, and then tried all SDTV-modes on CRT:

```
sdtv_mode=0          / Normal NTSC
sdtv_mode=2          / Normal PAL
sdtv_mode=16         / Progressive NTSC
sdtv_mode=18         / Progressive PAL
```

All modes works correct on CRT.

```
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 5.10.17-v7+ #1414 SMP Fri Apr 30 13:18:35 BST 2021 armv7l GNU/Linux
``` 

I'm still confused why lakka doesn't work, seems it has more fresh kernel.

gouchi, 2021-08-07:
So you did succeed to make CRT to work using KMS (no Xorg or Wayland) with RPiOS ?

Thank you.

eugene-s-nesdev, 2021-08-07:
I've just downloaded this "2021-05-07-raspios-buster-armhf.img" image, and tried all sdtv_modes (0, 2, 16, 18) via native RPi3 composite minijack output on CRT.
I'm not touched another lines in cfg.

gouchi, 2021-08-09:
Workaround for now is to use [fake kms](https://forums.libretro.com/t/lakka-lacks-crt-support/30424/26).

eugene-s-nesdev, 2021-08-09:
Does this config work with RPi3 internal minijack composite out?

jayare5, 2021-08-10:
> Does this config work with RPi3 internal minijack composite out?

It does!

eugene-s-nesdev, 2021-08-10:
ELZ wrote he use this config:

```
#include distroconfig.txt
dtoverlay=vc4-fkms-v3d
dtoverlay=dpi24
enable_dpi_lcd=1
display_default_lcd=1
dpi_output_format=519
dpi_group=2
dpi_mode=87
hdmi_timings=2048 1 180 202 300 240 1 3 5 14 0 0 0 60 0 42954545 1
disable_splash=1
disable_overscan=1
dtparam=audio=on
hdmi_drive=2
```

I see it contains a bunch of HDMI-related parameters.
What's clean config just for use with internal tvout?

This is not enough?:
```
dtoverlay=vc4-fkms-v3d
sdtv_mode=0
```

gouchi, 2021-08-21:
@jayare5 @eugene-s-nesdev Can you try to make a test  with [latest nightly](https://nightly.builds.lakka.tv/latest/) as we did [update the RPi kernel.](https://github.com/libretro/Lakka-LibreELEC/commit/6de244a0750e8820e4fd92e7443cc9a014ca37b3)

Thank you.

jayare5, 2021-08-23:
All right I've tested Pi3 and Pi4.

On Pi3, the situation is the same as before, needs to have the fkms edit in the config, and it will work properly with a CRT. Without the fkms it won't show up at all.

On Pi4, the situation has changed!
Using enable_tvout=1, no other edits, won't show up on the CRT.
Using enable_tvout=1, with the fkms edit, IT WILL SHOW UP ON THE CRT!
This is the first time I get that to happen on a Pi4 probably in the last 2 years.

Though I guess the point of updating is so that one day it will be able to work without the fkms edit?
Anyway, I'll be here to test future updates! :)

gouchi, 2021-09-02:
@jayare5  @eugene-s-nesdev 

Can you try for the RPi4 this configuration for `config.txt`

```
dtoverlay=vc4-kms-v3d-pi4,composite=1
hdmi_force_hotplug=0
hdmi_ignore_hotplug=1
enable_tvout=1
disable_fw_kms_setup=1
```

Then append to `cmdline.txt` the [video parameter](https://wiki.archlinux.org/title/kernel_mode_setting#Forcing_modes) and adapt to the resolution you need  `video=Composite-1:<xres>x<yres>@<refresh>e` for example 

`video=Composite-1:720x576@50ie`

Thanks to @HiassofT for the information.

jayare5, 2021-09-03:
Tested!

I flashed the nightly 2021 08 28, unfortunately it just hangs at the Lakka berry splash screen. (Likely same segfault as the logs I had posted)

I also don't think this line is doing anything:
video=Composite-1:720x576@50ie

I tried it in all 3 txt files just to see, and my TV always displayed things properly (480i @60, NTSC TV here but that line should have given me a PAL signal, which would have been wrong, but I tried it just like the example anyway and it doesn't do anything)

gouchi, 2021-09-03:
@jayare5 Thank you for your test.

So you did try also with  ?

`video=Composite-1:720x480@60ie`

May you try to give the output of when you are in the [terminal](http://www.lakka.tv/doc/Accessing-Lakka-command-line-interface/)

` for p in /sys/class/drm/*/status; do con=${p%/status}; echo -n "${con#*/card?-}: "; cat $p; done`

Thank you.

HiassofT, 2021-09-03:
make sure everything in cmdline.txt is on a single line, you have to add video=... after "quiet", separated by a space.

If you added that as a separate line it will be ignored.

jayare5, 2021-09-04:
Hey, it actually works! 
I put this line after "quiet " (with space)
video=Composite-1:720x480@60ie

It boots fine now and everything including cores seem to be quite smooth and perfect!
The only thing I don't see working now is sdtv_mode=16 for progressive NTSC, it's interlaced right now.

gouchi, 2021-09-04:
I am glad it is working now.

You may try to append the video parameter at the end of cmdline.txt with (without the **i**)

`video=Composite-1:720x480@60e`

HiassofT, 2021-09-04:
240/288 line progressive output isn't implemented in the driver yet.

There's an open PR though which should add this - when it gets merged

https://github.com/raspberrypi/linux/pull/4406

gouchi, 2021-09-04:
@HiassofT Thank you for the information I did not about it .

@jayare5 So we will have to wait a little ;)

HiassofT, 2021-09-04:
FYI: you can find some more info about composite configuration in the RPi kernel PR which added/fixed composite output: https://github.com/raspberrypi/linux/pull/4241

eg there's now also a vc4.tv_norm module parameter (which you can add to cmdline.txt) to select PAL60, SECAM, NTSC-J, ... output if you need that

jayare5, 2021-09-05:
Sounds like really good news!! I am glad to see this is just waiting to be merged and that a lot of attention is being put into this :D
Let me know when more tests are needed.

jayare5, 2021-09-08:
I tested the latest version after that merge!

Progressive certainly works, but not in the way I expected.
I think the only way for me to get progressive scan is to type `720x240@60e` in the parameter instead of `720x480@60e`?
However, the image is not correct because what that does is it takes the top 240 pixel portion and stretch it vertically... it needs to be 480, just "line halved" to 240 if that makes sense.

It is described here in the first paragraph titled Progressive modes: https://github.com/raspberrypi/linux/pull/4406 and apparently that's what is intended to do, but currently, it cannot be used like this.

So I just ask, is there / will there be a 480 mode that works how `sdtv_mode=16/sdtv_mode=18` did?
Thanks!!

kFYatek, 2021-10-03:
> It is described here in the first paragraph titled Progressive modes: raspberrypi/linux#4406 and apparently that's what is intended to do, but currently, it cannot be used like this.
As an author of that PR, I don't think that can be done directly.

With my PR, I was able to get the kind of output you expect in X by entering that 720x240 mode and running `xrandr --output Composite-1 --scale 1x2`. I'm not sure how could that be achieved in Kodi or whatever you're using.

The GPU will always internally operate at 720x240 in the progressive mode, that "fake" 720x480 mode you expect will always be some kind of scaling. Perhaps that could be configured within the DRM driver, but that's really outside my field or expertise or interest, and I'm not even sure if that kind of mode handling in the kernel would be appropriate.

jayare5, 2021-10-03:
Wow that sounds like bad news. That mode is specifically why many people like me even got into Raspberry Pi... it sounds like it will take very long for someone to come in with the right expertise and interest.
Things used to work perfectly well though, with just the `sdtv_mode=16` / `sdtv_mode=18`, is there no way to make that work without having to use `fkms` in `dtoverlay=vc4-fkms-v3d`? 
Because when I use fkms, certain things don't work well

kFYatek, 2021-10-04:
I just did some very dirty PoC on my Pi that does hardware scaling from 720x480 to 720x240 with Full-KMS. I did not test with RetroArch, but it seems to work under X, although not quite exactly like it does with the firmware framebuffer:
* each two lines of the 720x480 framebuffer are averaged instead of dropping every other one
* mouse cursor is a bit choppy - it only actually moves vertically on the screen after having been moved by 2 lines; i.e., its vertical position is rounded to the nearest physical pixel - this is due to it being a separate DRM plane; I don't know if software like RetroArch makes use of that or if that would be irrelevant

If this kind of behaviour would satisfy you, I can try polishing it up so that it could be made into an actual PR. I have no idea whether it could ever be accepted into the default Raspberry Pi kernel, let alone upstream - I don't think I've ever seen video modes downscaled by design in the kernel, so I wouldn't be surprised if the maintainers would be willing to accept that. On the other hand, that's how the firmware behaved since forever, so... maybe.

jayare5, 2021-10-04:
I really appreciate you continuing to work on this!! I did not expect it :) Thank you.
The way you describe the behavior does sound like it would work well for what we're doing here, because the thing is that the games we play with this are already basically line doubled (games are 240p, integer scaled to 480p, and then line halved again on the TV) so it shouldn't be a difference there! The lines being averaged sounds like it might even produce a nicer image whenever higher quality assets are shown (like in RetroArch menus).

If it's not too much to ask for, I'd love to try it!

kFYatek, 2021-10-04:
I've pushed my dirty PoC to my fork of the repo: it's tagged as [`480p-poc`](https://github.com/kFYatek/linux/releases/tag/480p-poc), the commit is [`8c4fc3c`](https://github.com/kFYatek/linux/commit/8c4fc3cda8695b0b583fc5be92521be0902375f9). It's based on the current state of that raspberrypi/linux#4406 PR.

It is intended to give you a clear "fake 480p" picture with `sdtv_mode=16`/`sdtv_mode=18`. **\*\*\*NOTE: It intentionally breaks all other modes, including HDMI\*\*\***, so proceed with caution.

> the games we play with this are already basically line doubled (games are 240p, integer scaled to 480p, and then line halved again on the TV) so it shouldn't be a difference there!

The reason why I was a bit skeptical is that, if an emulator displays the doubled lines at, say, y=0 and y=1, and then y=2 and y=3 and so on, there will indeed be no difference. But if instead it displays those doubled lines at y=1 and y=2, then y=3 and y=4 and so on - they will be _all_ blurred instead. I don't know if any emulator could align the image like that, it's just a possible corner case that I thought of.

> The lines being averaged sounds like it might even produce a nicer image whenever higher quality assets are shown (like in RetroArch menus).

This is probably true, though.

Additional note: with hardware scaling, it could also be possible to scale the image _up_, i.e. set something like 352x240 as the display mode, so that the aspect ratio is more or less correct, but there is no downscaling. I don't know if that would be interesting for you, just noting.

HiassofT, 2021-10-04:
There's no need to hack up the driver, DRM/KMS already supports scaling and you can set up framebuffers with different size than the CRTC output resolution if you want.

This gives you full control over what's happening in the output pipeline, automatic up- or down-scaling may not always be what you want (eg you might prefer to render to a 360x240 framebuffer and let DRM output that at 720x240).

Kodi uses that to up-scale the GUI to 4k on RPi4 and other SBCs (the GPU isn't powerful enough for smooth 4k GLES, so the GUI plane is limited to 1920x1080) and to scale videos to screen output size (eg playing 4k or SD content on a HD monitor).

kFYatek, 2021-10-04:
@HiassofT That's true - but the real question is whether RetroArch and other emulators support doing that.

If not, it might be actually easier to add implicit scaling to the driver instead of patching all the userland software, especially considering that it's how the firmware modes worked since forever...

HiassofT, 2021-10-04:
While it might be easier short term to carry over the old workaround it would be a lot better to implement that properly (if retroarch etc don't support it yet).

DRM will tell you the exact output resolution and should also report the picture aspect ratio which you can then use to calculate the output pixel aspect ratio and set up a proper scaling pipeline for emulator output so the picture won't be squashed or stretched.

While the usual PC/TV monitor modes nowadays have square pixels, that isn't true for PAL/NTSC modes (which could even be anamorphic 16:9) and even less true for 1980-era game consoles and home computers.

So to properly display emulator output on old and new monitors using correct aspect ratios a scaling pipeline has already to be set up and adding another step that involves lossy downscaling and is always happening in the background doesn't really help to get accurate output.

jayare5, 2021-10-04:
@kFYatek 

> The reason why I was a bit skeptical is that, if an emulator displays the doubled lines at, say, y=0 and y=1, and then y=2 and y=3 and so on, there will indeed be no difference. But if instead it [...]

Ah, I can tell you that this is the least of my worries :P RetroArch is full of video configs, moving the image 1 pixel up or down to make it align with the sharp line is the easiest thing to do! And it saves that location value so games always boot properly scaled / placed.

> Additional note: with hardware scaling, it could also be possible to scale the image up, i.e. set something like 352x240 as the display mode, so that the aspect ratio is more or less correct, but there is no downscaling. I don't know if that would be interesting for you, just noting.

That's good to know! But in reality i've found the downscaled approach to be more ideal because horizontal scaling needs to be higher res to stretch horizontally without noticing blurred pixels too much, like stretching 256 to around 640 or more.

> It is intended to give you a clear "fake 480p" picture with sdtv_mode=16/sdtv_mode=18. ***NOTE: It intentionally breaks all other modes, including HDMI***, so proceed with caution.

When you say break, does that mean that using HDMI could harm a TV? Or does it just simply mean I can use it and it will look like a weird 480p resolution?

Also, is there no way to ignore any of those sdtv commands like `sdtv_mode=16` / `sdtv_mode=18` (and maybe others) if HDMI is detected? Luckily I mainly use a CRT and rarely switch between it and HDMI, but I can put up with having to change the config.txt if I need to switch to HDMI for the time being.

@HiassofT 
> There's no need to hack up the driver, DRM/KMS already supports scaling and you can set up framebuffers with different size than the CRTC output resolution if you want.

Does this mean that HDMI does not have to be broken if using SDTV commands?

kFYatek, 2021-10-04:
> When you say break, does that mean that using HDMI could harm a TV? Or does it just simply mean I can use it and it will look like a weird 480p resolution?

No, it won't break a TV. By broken, I mean that all other modes will probably display a squashed image on only the upper half of the screen, or (less likely) some other form of distorted output. The signal itself should be reasonably formatted - just with garbage contents.

Besides, I don't think breaking an HDMI TV is that easy ;)

> Also, is there no way to ignore any of those sdtv commands like sdtv_mode=16 / sdtv_mode=18 (and maybe others) if HDMI is detected? Luckily I mainly use a CRT and rarely switch between it and HDMI, but I can put up with having to change the config.txt if I need to switch to HDMI for the time being.

I'm fairly sure that this is the case right now, with any "normal" kernel (i.e. not that PoC of mine I linked), at least with full-KMS. What problems do you encounter exactly?

jayare5, 2021-10-11:
@kFYatek 
So I was able to build an image (with the help of @dmrlawson, thanks a ton!!!!)
And I've finally tested the 240p stuff :D
It is perfect! I actually looked for those differences you mentioned where if it's 1 pixel off vertically, it would blur, it behaved like that so that is confirmation I tested your patch. And yes that "side effect" is absolutely fine, having line averaging is great, it does make the menu look a little better just like how I thought too! It does not affect the way games look at all, so this is good with me! I accept this as the new 240p behavior!

jayare5, 2021-10-11:
Truly appreciate all your work!!!! Thanks so much.

I will be testing each mode for sure this time! (I think the only mode that was usable when I tested was 480-scaled-to-240p)

Btw, what needs to happen for these to be accepted in the raspberrypi/linux?

jayare5, 2021-10-16:
@ToKe79 I tested the nightly builds, and it all seems perfect!!

I have a Pi3B+ and a Pi4. Both can now do 240p on a CRT and proper HDMI when editing the configs accordingly!
There are no glitches or issues as far as I could test on either board / config.
I can live live this now, unless there are plans in the future for both modes to work without having to edit the config?

Here are some notes just to make sure.
• On a **Pi3**, all I have to do is:

In cmdline.txt (after 'quiet ' with space included):
`video=Composite-1:720x480@60ie`

With this alone it will be 240p, and commenting that part out will work with HDMI.

---

• On a **Pi4**, I have to do the following:

In cmdline.txt (after 'quiet ' with space included):
`video=Composite-1:720x480@60ie`

In config.txt
`enable_tvout=1`
`dtoverlay=vc4-kms-v3d-pi4,composite=1`

(That second line is also in distroconfig.txt, but that file can be ignored apparently and only manage the config.txt for convenience)

Commenting out the part added to cmdline.txt will make it work with HDMI.

jayare5, 2021-10-16:
Looks like very clear instructions to me! :D Glad to see official support

[29 more replies]

closed 2021-10-17 by ToKe79