libretro/Lakka-LibreELEC #1297: RPi4.aarch64 - SYSTEM missing in nightly builds

rcasta74, 2021-05-16:
The nigthly build for RPi4.aarch64 is only 11M of size, and file SYSTEM is missing into the image.

It look like that the issue was introduced by PR 1278:
https://nightly.builds.lakka.tv/PRs/pr_1277_000291/RPi4.aarch64/Lakka-RPi4.aarch64-3.0-devel-20210425225330-4960a98.img.gz  (430M)
https://nightly.builds.lakka.tv/PRs/pr_1278_000293/RPi4.aarch64/Lakka-RPi4.aarch64-3.0-devel-20210426004732-f1cfdb4.img.gz    (11M)

Ntemis, 2021-05-16:
@dmrlawson any idea?

Ntemis, 2021-05-16:
I think image needs to be bigger to be able to include mame2015.

dmrlawson, 2021-05-16:
> I think image needs to be bigger to be able to include mame2015.

I think you're right, I thought it was disk space on the build box but it seems likely it's that the system partition is too small. Strangely it built for me last time. 

Should we disable something or make the partition bigger (again?)?

Ntemis, 2021-05-16:
No i prefer we make image partition larger and remedy this now before final release so we maybe avoid asking users in the future to start fresh.

closed 2021-05-16 by Ntemis