libretro/Lakka-LibreELEC #1301: Can't insatall on MIII-G 4k (S812)

Nevskiy83, 2021-05-17:
Cant't install lakka on my device. I burned image http://le.builds.lakka.tv/S8X2.MXIII-1G.arm/Lakka-S8X2.MXIII-1G.arm-2.3.1.img.gz with Etcher on SD card. Power on with reset button get black screen (without reset button black screen too). Now in NAND install openelec 6.0.0. I want install LAKKA. Can you help me? Thank you.

Ntemis, 2021-05-17:
https://le.builds.lakka.tv/S8X2.MXIII-PLUS.arm/Lakka-S8X2.MXIII-PLUS.arm-2.3.1.img.gz

Nevskiy83, 2021-05-18:
Thank you for answer. Unfortunately the result is almost the same. Now I have freeze on logo "Tronsmart". Power on button -> Logo "Tronsmart" -> black screen (change resolution?) -> logo "Tronsmart". This behavior with the reset button pressed and without

Ntemis, 2021-05-18:
Can you test a batocera image?

Nevskiy83, 2021-05-23:
> Can you test a batocera image?

I can't find batocera image fo s812 :(

ToKe79, 2021-05-23:
for the 3.0 image to work you have to edit the file `uEnv.ini` in the root of the FAT32 partition and set correctly `dtb_name=/dtb/@@DTB_NAME@@`. the device trees are in the `/dtb` subfolder. unfortunately, I do not know which one is suitable for your TV box. for example to use `gxbb_p200_1G_1Gbit.dtb`, then you make your edit like this:
```
dtb_name=/dtb/gxbb_p200_1G_1Gbit.dtb
```
leave the line with `bootargs=` as is.

ToKe79, 2021-05-23:
if you get stuck on the Lakka flower logo, you may add following at the end of the `bootargs=` line: `retroarch=0 textmode`. on the next boot you should be dropped to the shell and you can start [troubleshooting](http://lakka.tv/doc/Troubleshooting-Lakka/).

Ntemis, 2021-07-23:
S812 cant be supported on Jeos anymore. Feel free to come over to Batocera channel on discord for support and bug testing.
I am the one that did the port so feel free to ping me(Demetris)

closed 2021-07-23 by Ntemis