libretro/Lakka-LibreELEC #1302: Bluetooth enabled bug

Rstation4, 2021-05-19:
Raspberry Pi 4 2GB
Lakka version: Lakka-RPi4.arm-nightly-20210518-2bbd8bb

I just did a clean install of the Lakka image and when I went to settings -> Bluetooth it scanned even though the Services Bluetooth is not enabled. It did pick up one of my BT devices. I'm not sure if that's the norm for it or not since Ive only used BT once way back with a stable RPI2_RPI4  image.

ToKe79, 2021-05-21:
I suspect that when you go to the bluetooth menu it enables / turns on bluetooth. which I would consider as expected behaviour.

SethBarberee, 2021-05-22:
Yup, seems like it, @ToKe79 and @Rstation4. Might be good idea to push this over the Retroarch.

RPi uses bluetoothctl and powers it on, no matter what here:
https://github.com/libretro/RetroArch/blob/f8b6443b91ed5415fe82fdbd627c043d8c0d33f5/bluetooth/drivers/bluetoothctl.c#L52

ToKe79, 2021-05-23:
I would not consider this as bug, it is expected behavior - you go to the bluetooth menu to connect a bluetooth device, so it powers on the bluetooth adapter. of course, after reboot the service will not be running, as the service was not enabled.

closed 2021-05-23 by ToKe79

davidgfnet, 2021-05-24:
This probably needs a revamp like the wifi menu had. But yeah for now it's sort of working as intended

by-gnome, 2021-05-25:
@davidgfnet It would be great if you make a Bluetooth menu something like Wi-Fi menu :)