libretro/Lakka-LibreELEC #1309: Can beetle-saturn be re-added as a core?

Mr-Jake, 2021-05-21:
The core cleanup commit on March 25, 2021 dropped the beetle-saturn core from Lakka. This was a surprise because even though beetle-saturn has not been updated in a while, it has much better compatibility and performance compared to Kronos, especially for PC Generic x86_64.

I saw the recent commit that increased the partition size from 512mb to 2gb. If there is extra space now, could beetle-saturn be re-added even if only for Generic x86_64?

Ntemis, 2021-05-23:
Did you try yabasanshiro?

Mr-Jake, 2021-05-23:
I did not try yabasanshiro because I thought Kronos was an updated improved fork of it.

I may just be more of a fan of the beetle-saturn core. I could always just manually grab it from the buildbot and copy it to core folder in Lakka.

PR #1320 (2021-05-23): beetle-saturn: update and add the core back to list
was removed due to size constraints of the parition. as the partition
size was increased, the core can be included again.

fixes #1309
commit 372b3b6f (2021-05-23): beetle-saturn: update and add the core back to list (#1320)

was removed due to size constraints of the parition. as the partition
size was increased, the core can be included again.

fixes #1309
closed 2021-05-23 by Ntemis