libretro/Lakka-LibreELEC #1310: Any way to install packages

KaiH-0, 2021-05-21:
I have been trying for hours to add packages to Lakka but can't figure it out, I have tried multiple ways and can't get my head around recompiling it with packages installed and it doesn't seem that the nvidia version of Lakka can be compiled. I need to use apt but it isn't present which causes issues as I have some cores that aren't on retroarch and they need external programs to work.

If anyone has any ideas on how to sort this, it would be greatly appreciated if you could let me know.

gouchi, 2021-05-21:
Lakka is based on LibreELEC and there is no package manager.

We don't ship Nvidia driver as it is [not ready for DRM/KMS](http://us.download.nvidia.com/XFree86/aarch64/460.80/README/kms.html). The [nightly build](https://nightly.builds.lakka.tv/latest/Generic.x86_64/) is not working for your card ?

Which package do you need ?

KaiH-0, 2021-05-21:
I personally need quite a few so I would need to use any type of package manager to install them, some of them are custom programs.

gouchi, 2021-05-21:
Which packages do you need exactly ? For which purpose ?

Which card do you have ?

KaiH-0, 2021-05-21:
It's packages for a modified version of a core which requires the external program to be installed to use. My GPU is GTX 1060 3GB

gouchi, 2021-05-21:
You should add a package to libretro directory. Please check this [commit](https://github.com/libretro/Lakka-LibreELEC/commit/8ce48e6bd02ae789568303084c46e6dc2193503e) how to add a package.  Also check this [documentation](http://www.lakka.tv/doc/Packaging-a-libretro-core-for-Lakka/#forcing-rebuild-of-a-package).

Please provide some [logs](http://www.lakka.tv/doc/Troubleshooting-Lakka/) for the [nightly build](https://nightly.builds.lakka.tv/latest/Generic.x86_64/). Your card should work.

KaiH-0, 2021-05-21:
So any package can be installed like this? The documentation seems a bit confusing, but I'll give it a go.

gouchi, 2021-05-21:
Yes you may  check also this [README.md](https://github.com/libretro/Lakka-LibreELEC/blob/master/packages/readme.md).

KaiH-0, 2021-05-22:
I've managed to install opkg manager but I don't think that manager really lets you add repositories so I can't install the programs I need, is there anyway to install something like rpm, dnf, apt or something with it? Also thank you for helping me with this!

ToKe79, 2021-05-22:
@KaiH-0 unfortunately no - you cannot install anything in Lakka, as the rootfs is read only. you can however compile binaries for the target platform outside of Lakka and store them on the ext4 (/storage) partition, add the folder to PATH and it might probably work. The best method however is to "bake" what you need into the image by modifying the build system - e.g. adding a dependency to the package retroarch will install that package, as the build system will think that retroarch is dependent on that package. And the package you need must be in the `packages/` folder somewhere - if it is not there, you have to create the package yourself, or adapt existing package to your needs.

KaiH-0, 2021-05-22:
I've got it sorted for compiling it so I can do that it's just how to, I have tried cmake with apt but that didn't work.

ToKe79, 2021-05-22:
Are you trying to add package manager to Lakka? If so, this is not the right approach.

From what I read here, you are trying to add libretro-dolphin-launcher and dolphin standalone emulator to Lakka. If that is the case, you have to create packages for these - take inspiration from other packages already in the repository, as best base take any libretro-core package (as libretro-dolphin-launcher is a libretro core that, I assume, starts the standalone dolphin emulator) in `packages/libretro/`. Then you have to create a package for dolphin emulator and also for all its dependencies (great chance that for most dependencies of dolphin there is already a package). When you have done this, you just add libretro-dolphin-launcher to `LIBRETRO_CORES` in `distributions/Lakka/options` and dolphin as dependency of libretro-dolphin-launcher (of course, dolphin must have as dependency - PKG_DEPENDS_TARGET - all its dependencies). Do this for everything you want to have in the Lakka image. Then you create Lakka image and if everything goes well you will have your own custom Lakka with all tools and programs you want.

To be clear - Lakka is NOT a distribution. Lakka (based on LibreELEC) is "Just Enough OS" (JEOS) to do one single task with the least resources required/possible.

KaiH-0, 2021-05-22:
- What repo does it use when compiling that's the problem as it may not include dolphin. I have managed to make a opt directory in storage which has been used on the PATH so I can install packages, right now I have been trying to compile apt on it, so far it is nearly there the only thing that isn't working is BerkeleyDB as I can't get that installed in the moment as I am getting a ar: invalid option 'u' and apt requires BerkeleyDB, I have installed libdb47 with opkg and it doesn't recognise it which is why I am trying to compile the actual version of BerkeleyDB.

ToKe79, 2021-05-22:
@KaiH-0 You are making it the harder way I think - building a package manager directly in Lakka. As mentioned - you have to build an image that will include your stuff (emulators, helpers, scripts, ...) - you clone this repository and [build an image](https://github.com/libretro/Lakka-LibreELEC/wiki/Compiling-Lakka). When this is done, you can [add your packages](https://github.com/libretro/Lakka-LibreELEC/wiki/Packaging-a-libretro-core-for-Lakka#writing-the-packagemk) into the repository and try to build the image with these packages. If you want a package manager in Lakka, then Lakka is unfortunately not the best option for you.

ToKe79, 2021-05-22:
Also, I have to close this issue, as this is not a bug, but feel free to continue the discussion and ask questions. But maybe [Discord](https://discordapp.com/invite/27Xxm2h) would be better place for the discussion - in channel #lakkatv you will find many people that can help you.

closed 2021-05-22 by ToKe79