libretro/Lakka-LibreELEC #1316: Lakka 3 GPI Case build Raspberry Pi 0w wifi crashes system & screen rotated 90 degrees

michaelbradynet, 2021-05-23:
Lakka 3 GPI Case build Raspberry Pi 0w wifi network scan crashes system & screen rotated 90 degrees. These have existed in the nightlies before the 3.0 release as well. When you scan for wifi networks the systems freezes. The screen is also rotated 90 degrees clockwise .

Ntemis, 2021-05-23:
@SethBarberee is your guy

SethBarberee, 2021-05-23:
> Lakka 3 GPI Case build Raspberry Pi 0w wifi network scan crashes system & screen rotated 90 degrees. These have existed in the nightlies before the 3.0 release as well. When you scan for wifi networks the systems freezes. The screen is also rotated 90 degrees clockwise .

Screen rotation is a known issue from KMS migration. Not much we can do on the Lakka side until screen orientation is recognized. RA has an issue for that. (Trying to learn Mesa/KMS on side to fix this hopefully as Gnome has some prior art)

We do apply a rotation when you play the cores so it should just be the menu that this issue is occuring.

Wifi Scan is one that I'm trying to diagnose as I can reproduce it. Not sure when this popped up as I got my GPiCase rather recently. Trying to figure out the wifi driver more until I try to hack any fixes.

modeler, 2021-05-23:
Same issue here. I only tried the VICE Commodore 64 core, but the screen did not rotate.

Older nightlies had the correct screen rotation but would hang on the splash screen. This appears to be fixed with the addition of the following line in distroconfig.txt, where is when the rotation issue appears:

`dtoverlay=vc4-fkms-v3d,cma-128`

[image]

SethBarberee, 2021-05-26:
> Same issue here. I only tried the VICE Commodore 64 core, but the screen did not rotate.

Don't really use vice much but sounds like a core issue that it's not rotating. Does this happen for any other cores for you?

modeler, 2021-05-26:
Hi,

I tried the following:

Stella
Hatari
PrBoom
Beetle PCE
Gambatte
Genesis Plux GX
Snes9X 2002

None of which are rotated correctly when using Lakka-GPICase.arm-3.0.img.gz.

modeler, 2021-05-26:
If we remove `dtoverlay=vc4-fkms-v3d,cma-128` from **distroconfig.txt**, the orientation is correct but RetroArch refuses to launch and systemd goes into a loop:

[image]

SethBarberee, 2021-05-26:
That is expected. The display is initialized and ran of the firmware while Retroarch is driven by KMS. By turning off the overlay, you are disabling KMS support on the Pi.

SethBarberee, 2021-05-27:
> The splash screen is rotated also. I would be interested to know why display_lcd_rotate=1 no longer has any effect, did something change as part of a kernel upgrade? There was an older nightly that worked fine, I think from around October 2020. It is no longer on the buildbot server.

Didn't see this part but GPiCase and all other Pi's use the VC4 driver instead of broadcom closed source driver. `display_lcd_rotate` only works on `bcm2835` <- broadcom driver

modeler, 2021-05-27:
> That is expected. The display is initialized and ran of the firmware while Retroarch is driven by KMS. By turning off the overlay, you are disabling KMS support on the Pi.

That makes sense, thanks.

> Didn't see this part but GPiCase and all other Pi's use the VC4 driver instead of broadcom closed source driver. `display_lcd_rotate` only works on `bcm2835` <- broadcom driver

Did the older builds of Lakka use the Broadcom driver / no KMS? I would revert to v2.3.1 or the October nightly build, but I can't find either available for download now.

ToKe79, 2021-05-27:
old downloads are in https://le.builds.lakka.tv/legacy-2.3/

modeler, 2021-05-27:
> old downloads are in https://le.builds.lakka.tv/legacy-2.3/

Thank you, Lakka-RPi.GPICase.arm-2.3.2.img.gz works fine here.

SethBarberee, 2021-05-28:
Switching back to BCM driver for now as it seems not all cores support RA rotating them. Only GPiCase will be on the driver. All other Pi devices will still use VC4 though. Will get a PR up shortly

modeler, 2021-06-06:
Sorry to be a pain but where might I find the BCM driver build for the GPiCase please? If it's not done yet, no worries.

I tried the recently released Lakka 3.1 but it suffers from the same issues as 3.0 with this device. Thanks!

SethBarberee, 2021-06-06:
Was on vacation but the WIP PR is here:

https://github.com/libretro/Lakka-LibreELEC/pull/1346

Currently fixing build issues before it's good for testing.

SethBarberee, 2021-06-08:
@modeler @michaelbradynet 
Most recent build from that PR can be used if you want something now. It has been merged but after the nightly started so 2 nightlies, it'll be there. With the switch, Wifi doesn't crash anymore, and of course, rotation settings from `distroconfig.txt` are honored since it's using firmware/broadcom driver. Lakka 3.2 will have the change too.

modeler, 2021-06-09:
I tried the image Lakka-GPICase.arm-3.0-devel-20210608233405-7af0ecf.img.gz but the screen is still rotated as before.

john-parton, 2021-06-17:
@modeler Use the latest build (`Lakka-GPICase.arm-nightly-20210615-2a94fae.img.gz`) and it seems to work correctly.

modeler, 2021-06-17:
That works perfectly, thanks all!

closed 2021-06-18 by ToKe79