libretro/Lakka-LibreELEC #1319: Is it possible to install Lakka on different distro?

KaiH-0, 2021-05-23:
I know that Lakka has a modified version of retroarch, would it be possible to get that version of retroarch and use it on a different distro?

ToKe79, 2021-05-23:
you can compile retroarch by adding `HAVE_LAKKA=1` to the make options, i.e. run `./configure && make HAVE_LAKKA=1 HAVE_BLUETOOTH=1 HAVE_ZARCH=0` as done [here](https://github.com/libretro/Lakka-LibreELEC/blob/359ef661c4a94e0f3be398a813828892cc7b6a14/packages/libretro/retroarch/package.mk#L162).

ToKe79, 2021-05-23:
But you will not have access to the core downloaded, so you might need to patch the source as well, so you get the core downloader

closed 2021-05-23 by ToKe79

KaiH-0, 2021-05-23:
Where do you do the ./configure, I see a makefile but nothing for configure.

ToKe79, 2021-05-23:
you download the RetroArch source / clone the repository. you run the `configure` script and then run `make` with desired options.