libretro/Lakka-LibreELEC #1328: Atomiswave Gets Stuck on System Settings Menu (RPI4) (Flycast)

Rstation4, 2021-05-24:
Raspberry Pi 4 2gb
Lakka Nightly Version for RPI 4

I already tried this with a different awbios.zip but it still happens when I try to exit the system settings menu. I tried this with another rom as well with the same result of it freezing up Lakka OS when I try to close the rom. 

 Why does the "Allow Naomi Service Buttons" option need to be enabled to use Atomiswave System Settings? I wasn't able to do so unless I enabled that in the menu
 
 Logs:
 
[Atomiswave lakka os.log](https://github.com/libretro/Lakka-LibreELEC/files/6530886/Atomiswave.lakka.os.log)
[Atomiswave peformance mark.log](https://github.com/libretro/Lakka-LibreELEC/files/6530887/Atomiswave.peformance.mark.log)

 image:
[image]

gouchi, 2021-05-25:
Hi,

You did try to use L3 (Test) and R3 (Service) ?

Rstation4, 2021-05-26:
Yes I did, while the rom ends up freezing when you try to exit, the whole Lakka OS will freeze when you try then try to close it via close content.

Rstation4, 2021-05-30:
@gouchi I moved on to the Lakka OS 3.0 Stable Release and something there fixed. It has not happend since then, though I have only tested it on 4 roms. Should this issue be considered fixed and closed?

ToKe79, 2021-05-30:
please reopen when needed

closed 2021-05-30 by ToKe79