libretro/Lakka-LibreELEC #1332: Since Mupen6Plus has been removed in 3.0, N64 games don't start at all

x2k13, 2021-05-25:
Lakka Version:
Fresh installation of 3.0

Hardware: 
Raspberry Pi 3B+

As the title says, since the removal of Mupen64plus in favor of Mupen64Plus-Next in 3.0, N64 games don't start anymore.

What happens?
Starting a game just leads to a force-close and throw-back to "desktop".

What did you expect to happen?
The game should start as usual, on 2.3.2, when using Mupen64Plus-Next

Rolled back from 3.0 to 2.3.2 for the time being, so if logs are needed I'm going to re-flash 3.0

gouchi, 2021-05-25:
They are important ;)

> It is important to provide logs for debugging as part of your report. Instructions for generating logs can be found in the Troubleshooting Lakka documentation at https://www.lakka.tv/doc/Troubleshooting-Lakka/

Thank you.

x2k13, 2021-05-28:
> They are important ;)
> 
> > It is important to provide logs for debugging as part of your report. Instructions for generating logs can be found in the Troubleshooting Lakka documentation at https://www.lakka.tv/doc/Troubleshooting-Lakka/
> 
> Thank you.

Somehow I knew you'd say that. ;) 

Okay, here we go:
[https://vim.cx/?b65ce794b5f0fc8d#4wxehkBDEZ3QwCn8cpc6s5oTT9WhjiUPob6ovagFU8Ed](url)

I hope that helps. :)

x2k13, 2021-06-23:
No change in 3.1, still no working N64 EMU on lakka on RPi 3B+.
ParaLLEI N64 is producing, if not instantly crashing, a slideshow resulting in a full system freeze.
Mupen64plus-Next just crashes like on 3.0.

Fresh installation of 3.1 of course...

dmrlawson, 2021-07-06:
I made a bug for the core.

It's possibly a bug with mesa, but since we won't want to downgrade it we can only hope to narrow it down and make an issue there or wait and have it magically fixed.

x2k13, 2021-07-07:
> I made a bug for the core.
> 
> It's possibly a bug with mesa, but since we won't want to downgrade it we can only hope to narrow it down and make an issue there or wait and have it magically fixed.

I'll be using 2.3.2 until then. 3.x is really no improvement on my end.

closed 2021-07-23 by Ntemis