libretro/Lakka-LibreELEC #1337: Neo Geo AES Bios does not work on Rom (RPI 4)

Rstation4, 2021-05-26:
Raspberry Pi 4 2gb
Lakka OS Lakka-RPi4.aarch64-3.0

Core: FBNeo
Rom: Metal Slug 3
Rom Format: zip

Rom does not work at all when using AES bios option "AES Japan" and "AES Asia". I get a blank screen that flickers with purple and white. I did test this out with another mslug3.zip rom only for  the results to be the same.

Log:
[NeoGeo AES bios.log](https://github.com/libretro/Lakka-LibreELEC/files/6543342/NeoGeo.AES.bios.log)

Img:
[image]

jdalmanza, 2021-05-26:
first, what romset are you using? Are you sure it is verified compatible with the latest fb neo? Also the neogeo.zip must be from the same romset
Second, you really should be using the unibios if you are changing the bios from the default at all. unibios 4.0 is free now and so much better than the stock bios.
FYI I can verify that unibios works just fine on lakka 3.0 aarch64 on pi4 with fbneo I literally just finished metal slug x this evening :-)

Rstation4, 2021-05-26:
I've heard different from 2 other people that uni-bios is known to have bugs. I do have other Uni-bios versions besides 4.0. I have tested this out on Metal Slug 4 as well and there was some weird issues of console mode not allowing to jump,shoot or use grenades. When I set the Unibios to run the the rom of Metal Slug 3 in console mode as well I get that issue as well with not being allowed to use the actions. 

The romset did not come with neo geo bios, not the first time since i have seen 2 other rom sets without the neogeo.zip bios

I reported that here: https://github.com/libretro/FBNeo/issues/777

closed 2021-07-23 by Ntemis