libretro/Lakka-LibreELEC #1340: Lakka 3.0. Orangepi plus2e. Wifi doesn't work.

zarkon0, 2021-05-26:
With lakka v. 3.0 wifi doesn't work in Orangepi plus 2e.
The wifi doesnt find any network.

davidgfnet, 2021-05-27:
Do you know how to ssh into your device?
Could you please check whether the wifi works on Linux? 
You could run "ifconfig -a" and check whether there's a wifi interface. That could help to udnerstand whether we have a Linux issue or a Retroarch issue.
Also dmesg and "lsusb" outputs would help.
Thanks!

ToKe79, 2021-05-31:
Hi @zarkon0,
can you check community image from [here](https://nightly.builds.lakka.tv/members/vudiq/Lakka-LE_master/H3.arm/)?
It is based on LibreELEC master branch, Lakka 3.0 is based on the LE9.2 branch.

zarkon0, 2021-05-31:
Hi.

I can't connect with ssh because I can't connect with ethernet wire.
Can I acess to linux console stand alone in lakka?

I'm to try a nightly build.

Thanks.

zarkon0, 2021-05-31:
Hi again.

I've install Lakka-H3.arm-LE_master-devel-20210530212829-152e433-orangepi-plus2e.img.gz and the wifi doesnt' work.

The menu shows the driver connmanctl but if I press the button to change only null appears and the item connmanctl doesn`t
show checked.

This happens in 3.0 and nighty versions.

ToKe79, 2021-05-31:
Can you test an image from LE directly - https://test.libreelec.tv/.
If it does not work, you have to check for issue / open new issue in their repo.

diegohvanni, 2021-06-03:
I have a similar problem with Orange pi lite. The wireless module works properly on Lakka 2.3.1, retrorangepi and armbian.

Ntemis, 2021-06-30:
RTL8189ETV is not supported yet by any kernel.

closed 2021-06-30 by Ntemis