libretro/Lakka-LibreELEC #1342: RG351P Image - Request to add Support for RTL8811AU WiFi Adapters

rstrube, 2021-05-26:
Unfortunately the RG351P devices do not come with built in WiFi (_note: technically the first batch of RG351P devices came with internal WiFi_) so you need to use a USB WiFi Adapter. Many folks get a WiFi adapter based on the RTL8188EUS chipset, unfortunately I have a the TP-Link AC600 model which uses the RTL8811AU chipset. Other firmwares for the RG351P devices _do_ support this WiFi adapter so I was hoping that support could added in Lakka.  Based on my investigation, support for this particular WiFi chipset is out of tree and is typically built using DKMS (although I must admit I could be missing something).

Would it be possible to build out of tree kernel modules and include them in official Lakka images? If so, could support for the RTL8811AU chipset be added to the Lakka image for the RG351P?

Thanks for any assistance and/or advice you can provide!

- Which version of Lakka are you using?
3.0

- What system hardware are you using?
RG351P

- What did you do?
Attached a WiFi Adapter using the RTL8811AU chipset.

- What did you expect to happen?
WiFi Adapter recognized by Lakka

- What happened instead?
WiFi Adapter not recognized.

gouchi, 2021-05-26:
It seems to be [disabled](https://github.com/libretro/Lakka-LibreELEC/blob/master/projects/Rockchip/devices/OdroidGoAdvance/options#L59) as it is not working properly ? But indeed it is not defined in Lakka options, so you we may try to enable it for OGA devices.

Actually the [repository](https://github.com/MilhouseVH/RTL8812AU) does not exist anymore.

Can you provide `lsusb` output ?

Maybe we will have to use the [one used by LE](https://github.com/LibreELEC/LibreELEC.tv/blob/master/packages/linux-drivers/RTL8812AU/package.mk#L9) or this  [repository](https://github.com/abperiasamy/rtl8812AU_8821AU_linux) instead ?

Thank you.

rstrube, 2021-05-26:
Here is `lsusb -v` output taken from my PC with the WiFi adapter connect (so the bus IDs won't make sense).

```
Bus 002 Device 006: ID 2357:011e TP-Link AC600 wireless Realtek RTL8811AU [Archer T2U Nano]
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.10
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x2357 TP-Link
  idProduct          0x011e AC600 wireless Realtek RTL8811AU [Archer T2U Nano]
  bcdDevice            2.00
  iManufacturer           1 Realtek 
  iProduct                2 802.11ac WLAN Adapter 
  iSerial                 3 00e04c000001
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x003c
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           6
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol    255 Vendor Specific Protocol
      iInterface              2 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x05  EP 5 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x06  EP 6 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x87  EP 7 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               3
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x08  EP 8 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x09  EP 9 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
```

rstrube, 2021-05-26:
It appears most information I can find online references the following drivers:

https://github.com/aircrack-ng/rtl8812au (Arch Linux uses this driver to support rtl8811au, rtl8812au, rtl8821au chipsets)
https://github.com/gnab/rtl8812au (this seems to be recommended to Raspbian users to support rtl8811au, rtl8812au, rtl8821au chipsets)

I think in both cases the driver support all three chipsets.

davidgfnet, 2021-05-26:
Ah weird. My 351P did not ship wifi but I manually installed it. However I thought they were using MT7601U, so thats what I enabled.
No worries we can enable as many usb wifi chips as we want, to improve compatibility on these devices. Will also benefit all.OGA users that use random ebay dongles.
I will send a PR soon to fix this. Thanks!

davidgfnet, 2021-05-27:
So it looks we need an out-of-kernel driver for this? I can't find this driver upstream.
Are other distros using the out of tree one too?

rstrube, 2021-05-29:
@davidgfnet I believe that's correct.  It is out-of-kernel, built using DKMS.  The other distros for the RG351P (ArkOS and 351ELEC) build this driver into the image.

It appears that Lakka does _already_ reference the out-of-kernel driver [already](https://github.com/libretro/Lakka-LibreELEC/blob/master/packages/linux-drivers/RTL8812AU/package.mk#L9)

Per @gouchi 's comment, it also appears that those specific drivers for the OGA images (351P uses this same image) have been [disabled](https://github.com/libretro/Lakka-LibreELEC/blob/master/projects/Rockchip/devices/OdroidGoAdvance/options#L59).  Not sure why the drivers have been disabled for OGA images in particular.

[The LE linux driver](https://github.com/LibreELEC/LibreELEC.tv/blob/master/packages/linux-drivers/RTL8812AU/package.mk#L9) uses: https://github.com/aircrack-ng/rtl8812au

[The Lakka linux driver](https://github.com/libretro/Lakka-LibreELEC/blob/master/packages/linux-drivers/RTL8812AU/package.mk#L9) uses: https://github.com/MilhouseVH/RTL8812AU

[The 351ELEC linux driver](https://github.com/351ELEC/351ELEC/blob/main/packages/kernel/linux-drivers/RTL8812AU/package.mk#L9) _also_ uses: https://github.com/MilhouseVH/RTL8812AU

rstrube, 2021-06-05:
@davidgfnet I was wondering if you had a chance to look into re-enabling these WiFi drivers for the OGA images? Thanks for your help and assistance!

davidgfnet, 2021-06-05:
Sounds like quite complicated and i dont like the fact that they are not supported upstream.
But I guess theres many drivers in this situation.
Perhaps we should have a wiki page with all the compatible wifi usb dongles so ppl dont buy the wrong one.

rstrube, 2021-06-05:
@davidgfnet it looks like support for these out of tree drivers has already been wired up within Lakka, it's just been explicitly disabled for the OGA images.

Here is the package.mk file already within the Lakka source tree: https://github.com/libretro/Lakka-LibreELEC/blob/master/packages/linux-drivers/RTL8812AU/package.mk#L9

And here is where those additional drivers are removed for the Odroid Go Advance image: https://github.com/libretro/Lakka-LibreELEC/blob/master/projects/Rockchip/devices/OdroidGoAdvance/options#L59

My understanding is that these lines:

```
# Remove some broken drivers
ADDITIONAL_DRIVERS="${ADDITIONAL_DRIVERS//RTL8192CU/}"
ADDITIONAL_DRIVERS="${ADDITIONAL_DRIVERS//RTL8192EU/}"
ADDITIONAL_DRIVERS="${ADDITIONAL_DRIVERS//RTL8812AU/}"
```

Essentially remove RTL8192CU/EU and RTL8812AU from the `ADDITIONAL_DRIVERS`  variable, meaning that those drivers are _not_ built as part of the Odroid Go Advance image.  I believe simply removing these lines will mean that those drivers will be built again as part of that image.  In fact, those drivers are probably already being built for _other_ devices images, just not the OGA images.

The real question is why have those drivers been turned off for the OGA images?  They shouldn't be loaded unless users are explicitly using those particular WiFi adapters.

Ntemis, 2021-06-06:
Is looks crystal clear to me by the comments 😅

shantigilbert, 2021-06-06:
https://github.com/EmuELEC/EmuELEC/blob/dev/projects/Rockchip/devices/OdroidGoAdvance/packages/RTL8812AU/package.mk

This package is the one I use in EmuELEC and it works.

davidgfnet, 2021-06-06:
Hey wait a moment. In the title of the issue you talk about RTL8811AU, but then you refer to the RTL8812AU driver (note the +1 in the numbering!). What driver do you refer to really?
Also you are right that OGA disables that driver explicitly but there seems to be a good reason for it: the kernel it ships already has it. I found both this:

```
projects/Rockchip/devices/OdroidGoAdvance/options:    ADDITIONAL_DRIVERS="${ADDITIONAL_DRIVERS//RTL8812AU/}"
```

That removes the out of tree driver but also this:

```
projects/Rockchip/devices/OdroidGoAdvance/linux/odroidgoA-4.4/linux.aarch64.conf:CONFIG_RTL8812AU=m
```

And based on my build output, there's a driver already for it in OGA:
/usr/lib/kernel-overlays/base/lib/modules/4.4.189/kernel/drivers/net/wireless/rtl8812au/rtl8812au.ko

Is it that the driver is just not working? Not sure why that kernel tree has the driver, perhaps was it later removed? Probably Hardkernel merged it for compatiblity?
No idea, but anyway it's there and should work. Can you provide dmesg, lsusb, ifconfig/iwconfig outputs to better diagnose the issue?

rstrube, 2021-06-08:
@davidgfnet Sorry for my delayed response.

All these model numbers are confusing! But I believe in this case the driver for the **RTL8812AU** _also_ works for the **RTL8811AU**.

I will setup a fresh SD card with Lakka 3.1 (hopefully tonight) and provide the additional information you've requested.  Thanks for your assistance, it's very much appreciated.

closed 2021-07-26 by Ntemis