libretro/Lakka-LibreELEC #1352: PPSSPP Core: Movie Intro's lag and crash (RPi 4) (Lakka 3.0 stable)

Rstation4, 2021-05-30:
Raspberry PI 4 2gb
Lakka 3.0 Stable Version for RPi 4 arm.
Core: PPSSPP
Rom format: iso

In the prior Lakka Versions and Nightly versions for the RPi 4 I have noticed that PPSSPP has had some issues. The intro's tend to  lag, which I had seen with plenty of roms for the PSP system. With this recent Lakka version I have tried several roms where the issue occurs as well such as "Metal Gear Acid (USA)", "Summon Night 5 (version 2) (USA)", "Metal Slug Anthology (USA)" and "Jeanne d' Arc (USA)".  With the case of "Metal Slug Anthology" The intro lagged and then it froze the RPi 4 unit hard at the end of the intro which I needed to turn off the unit" 

It kept repeating some last 2 lines and the original log bloated up to 140 Mb's. 

log:
[Retroarch ppsspp.log](https://github.com/libretro/Lakka-LibreELEC/files/6566032/Retroarch.ppsspp.log)

video clip:

https://user-images.githubusercontent.com/73009731/120100774-6a6e4a00-c0f7-11eb-8d2a-ebbba96030fa.mp4

Ntemis, 2021-05-30:
It's occuring in all retro distros. The issue is mainly on ffmpeg. You should report this to the core developers as it's not lakka related

Rstation4, 2021-05-30:
I reported it there  https://github.com/hrydgard/ppsspp/issues/14491 
Should I close this one now?

Ntemis, 2021-05-30:
Lets keep it open so we track when issue is fixed

gouchi, 2021-05-31:
> If you're using system FFmpeg and a PPSSPP stable version, this is not performance related but because you're using system FFmpeg (which we never recommend.) However, this issue has been improved in latest git (of course future FFmpeg versions may break it again, and we still don't recommend system FFmpeg.)

What was the issue with using [FFmpeg provided in PPSSPP source](https://github.com/libretro/Lakka-LibreELEC/blob/master/packages/libretro/ppsspp/package.mk#L50) (changing to `-DUSE_SYSTEM_FFMPEG=OFF`)? It does not compile on some platform ?

Ntemis, 2021-05-31:
We can certainly try but I as working fine before

gouchi, 2021-06-02:
@Rstation4 Is it working better with #1364 ?

Thank you for your test.

ToKe79, 2021-06-02:
What I recall - we switched to the system ffmpeg because the commit used by ppsspp at that time was very old/outdated and we already build ffmpeg for the RetroArchs movie player (?), so why not use it for ppsspp as well...

unknownbrackets, 2021-06-03:
The problem is that PSP games can use a PSP API to decode videos which works pretty differently from FFmpeg.  They can add streams via an API call, even if the video header doesn't have them - and PSP games need this to work.  Newer versions of FFmpeg have been modifying the FFmpeg APIs to be a bit more foolproof to use, but don't allow streams to be added anymore.

The version of FFmpeg in ppsspp-ffmpeg isn't patched in any way (anymore, it used to be), it's just an old version that works well.  The formats PSP games support haven't really seen any updates in FFmpeg for years since that old version.  Latest git builds have a workaround for the latest FFmpeg, but some games may have bugs that don't normally happen.

-[Unknown]

ToKe79, 2021-06-03:
Thanks @unknownbrackets!
So if we stick to the ffmpeg submodule in ppsspp we should be on the safe side? Or should we build ppsspp-ffmpeg with latest commit? we use 1.11-hotfixes branch, but also master checks out hrydgard/ppsspp-ffmpeg@0b28335acea4f429ae798c5e75232e54881bf164.

unknownbrackets, 2021-06-03:
Yeah, that should be safe.  Building ffmpeg will take a bit longer but is likely the best option.

We initially intended to include only Windows and Android binaries for people building on Windows, because FFmpeg used to be quite complex for the average developer to build on Windows.  It's still complex, although much better.

On Linux and etc., though, it's relatively straight forward, so I think it's better to just build it each time.  Less risk of mismatching compiler ABI settings or etc.

-[Unknown]

Ntemis, 2021-06-06:
@Rstation4 test 3.1 release

Rstation4, 2021-06-07:
I went ahead and tested it after updating my Lakka version to a 3.1 rpi4 (aarch64) stable version.

I did a short test with a few roms and I did not see that issue. But as for the core as a whole I did not do any gameplay testing yet. I have a log generated as well. I dont have time this morning to do further gameplay testing. I did enable a performance marker option for the logs, but I don't see it generated, are they now combined in the general Retroarch.log now?

logs:
[retroarch.log](https://github.com/libretro/Lakka-LibreELEC/files/6608756/retroarch.log)

ToKe79, 2021-06-10:
there were no other changes to the core, except linking with non-system ffmpeg. imho positive fmv test is sufficient for this issue to be considered as fixed. if there are other issues, feel free to open a new issue.

closed 2021-06-10 by ToKe79